iceman1001/ChameleonMini-rebooted #1: rev E rebooted ==  RDV2.0?

ikarus23, 2017-11-27:
Hi @iceman1001,
maybe it's a stupid question, but is the [rev E rebooted](https://lab401.com/products/chameleon-mini-reve-rebooted?variant=50258105543) the same as the [RDV2.0](https://de.aliexpress.com/item/Newest-13-56MHZ-ISO14443A-Chameleon-Mini-RDV2-0-Kits-RFID-Copier-Duplicator-UID13-56MHZ-NFC-Card/32820382051.html?ws_ab_test=searchweb0_0,searchweb201602_2_10320_10152_10321_10065_10151_10344_10068_10345_10342_10547_10343_10322_10340_10341_10548_10193_10194_10541_10562_10084_10083_10304_10307_10302_10539_10180_10312_10059_10313_10314_10184_10534_10533_100031_10319_10604_10603_10103_10186_10594_10557_10596_10595_10142_10107,searchweb201603_25,ppcSwitch_5&btsid=0143828e-c603-492e-96c5-048b872feede&algo_expid=efd67f9d-5b9e-4375-a711-da7c06565098-0&algo_pvid=efd67f9d-5b9e-4375-a711-da7c06565098&rmStoreLevelAB=4)?

WhityKun, 2017-11-27:
And a following stupid question - is there a pcb files for rev E rebooted?

iceman1001, 2017-11-28:
@ikarus23  Not a stupid question.  Very confusing all shops selling with different names but to answer the question.  Yes,  I think so.  The pcb looks the same.

@WhityKun   No,  I didn't get any pcb files.

ikarus23, 2017-11-28:
@iceman1001, thanks for the reply! This device is definitely  going on my shopping list!

closed 2017-11-29 by iceman1001