iceman1001/ChameleonMini-rebooted #2: build failure.

iceman1001, 2017-11-28:
Getting the project to build.

user@ubuntu:~/ChameleonMini-rebooted/Firmware/Chameleon-Mini$ make
Makefile:120: ../LUFA/Build/lufa_core.mk: No such file or directory
Makefile:121: ../LUFA/Build/lufa_sources.mk: No such file or directory
Makefile:122: ../LUFA/Build/lufa_build.mk: No such file or directory
Makefile:123: ../LUFA/Build/lufa_cppcheck.mk: No such file or directory
make: *** No rule to make target '../LUFA/Build/lufa_cppcheck.mk'.  Stop.

bogiton, 2017-11-28:
Maybe try and get them from: https://github.com/abcminiuser/lufa/tree/master/LUFA/Build ?

ceres-c, 2017-11-28:
Trying to work on it, initial commit on my fork
I don't know if I'll be able pull this out in 30 minutes, later I have to go and can't work on it today

iceman1001, 2017-11-28:
Nice one, bro!  
I got rid of some warnings too.

ceres-c, 2017-11-28:
I'm working on getting it to compile, but I've got a doubt. Given we're working with a chinese codebase different from the original one and quite messy (we've been given a codebase not compiling and that's enough), I don't know what to do.
The file AntennaLevel.c is missing and AntennaLevel.h is quite similar but our version miss 2 functions which are present in the main repo, naming
```
void AntennaLevelTick(void);
void AntennaLevelResetMaxRssi(void);
```

Now, should I copy-paste (brutal, I know) the official AntennaLevel.c and update AntennaLevel.h accordingly or should I strip out unused functions from the c file?

iceman1001, 2017-11-28:
Go nuts,   I already did merging with revG,  as mentioned in issue 3 ...     Mea culpa.

This codebase is old,  so first I tried to get official revE merged,  and  thought that the changes should be easy to spot the difference but they were plenty.   

So either we go nuts or we start over from scratch,  merging offical revE with latest fixes on to this one.
What do you think?

ceres-c, 2017-11-28:
Kinda hard to decide, that's why I asked. We might go for the easy way and hack it together, but it might make thing way worse later. Actually I have no idea how the chinese managed to use this, given how many errors there were just for compilation...

BTW, funny thing: I don't own the original chamelon, nor this one. Just doing it for fun, but I can't test anything. Hehe

iceman1001, 2017-11-28:
The biggest advantage is the GUI and the reader-attack for mifare classic. 
and eight leds.  ...  lets go nuts and hack it up :)

We can always start fresh from the original codebase with all we learned from this practice.

No chameleon?!?  And you call yourself a pentester ?  ;)  just kidding.  Its chrismas soon!

ceres-c, 2017-11-28:
So we're going to hack our way into new features. This is going to be fun to watch :-D

Yeah, this Christmas I'll gift myself some stuff, you bet it ;-) I had a 180$+ cart on rfxsecure, that was only tags eheheh... Still I believe I'll go for the official version with chameleon

ceres-c, 2017-11-28:
Also, @iceman1001, do you have keybase or something like that so I can contact you via chat? That'd be helpful given I don't have a chameleon to use for tests and I'd do with some more information in order to work on this :-/

iceman1001, 2017-11-29:
It took some years before I got one aswell.  Wanted to know what the fuzz was about. 
Signal?  irc?  open thread on forum?

ceres-c, 2017-11-29:
Yeah, I got into RFID quite late... I'm no Major Malfunction, after all

I'd go with signal given I already have it installed on my phone and I don't have an IRC client to use "on the go". Won't spam you, I promise. Also, the forun doesn't seem too suit given I might have to ask you info to understand this program but mostli non-technical stuff which isn't worth getting everybody an alert ;)

gtpy, 2017-11-29:
I'm new into NFC, but like reversing etc.
I have an original chameleonmini revG.
Next investment will be a Proxmark3.
Mabe I can contribute with some Infos or check out the GUI

iceman1001, 2017-11-29:
Go for it!  I just want this project to compile as-is now.

iceman1001, 2017-12-14:
Yes!   It compiles!    Woop!

closed 2017-12-14 by iceman1001