iceman1001/ChameleonMini-rebooted #6: more utils

iceman1001, 2017-11-29:
During the short time I have had time to play with a chameleon mini, I noticed an sense of missing utils.

Its like PM3,   most stuff is there but you need to know where they are.  Not super clear but there.

With Mini its not there.  There is a python client,  which I still haven't used.  
So where is all the dump-utils,  you know, where you convert from different sources like MCT,  PM3, LibNFC,   all have EML, BIN, MFD  which is kind of the same but with minor differences.
And which dump formats is supported?!   For a device which uses dumps,  I see very little usability for endusers.

- dump converters
- dump + key mergers
- dump viewers

Some of these exits for PM3, which we can just copy over,  but still..

Which utils are you missing?

kevin2008-01, 2017-12-01:
Hi,

.dump to .bin or .eml converter please.

Just rename the file ?

Thanks

kevin2008-01, 2017-12-01:
Here, is the software pack with my chameleon.

http://uptobox.com/gyk7l878jzd5  if you want see.

iceman1001, 2017-12-01:
Great! 
Still such a raw enduser experience.  Coming from Proxmark3, I thought Chameleon Mini would be _kinder_

kevin2008-01, 2017-12-01:
yes iceman. i explore mf detect... but no success.... how to ?

iceman1001, 2017-12-01:
mf detect?

kevin2008-01, 2017-12-01:
is't possible add to gui, increment -1 / +1 same original ? or script, or....

kevin2008-01, 2017-12-01:
MF_DETECTION Fonction with reader attak

iceman1001, 2017-12-01:
aha.  I see.  No idea,  since I don't have acces to the GUI source code. :(

kevin2008-01, 2017-12-01:
Here,

The original mini rev G gui software.
http://uptobox.com/dkr4dgwdmsbz

iceman1001, 2017-12-02:
Some dump converters from proxmark3 repo.    Would need dump2eml aswell.

https://github.com/iceman1001/ChameleonMini-rebooted/commit/db0e443d4b0905b136a3156e595121b813776162

kevin2008-01, 2017-12-03:
thank you for:
https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Software/Tools

kevin2008-01, 2017-12-14:
Merci, i see day by day this gith. Tank you very much.

WolfgangMau, 2018-01-20:
I hacked me a little [terminal-wrapper](https://github.com/WolfgangMau/chamgo) in golang
it's nothing special, but at least it works on osx, linux & windows in the same way
the folder `compiled` contains precompiled binaries
tested on osx (sierra) & windows7

iceman1001, 2018-01-20:
Unfair,  golang is so much simpler ;)  
Great addition,  wasn't a terminal wrapper in python before?!

WolfgangMau, 2018-01-20:
don't know ... I was just bugged while switching between windows and osx ;-)
and I try to setup at less as possible on windows (e.g. pythons or buldchain)
so laziness wins and I choosed golang

iceman1001, 2018-01-21:
I notice the tools from RevG is not here,   lets see if we can add your repo also..
All those tools are commandline-based?   I wonder if the GUI can be made in Go aswell?

ref:
https://github.com/emsec/ChameleonMini/tree/master/Software

WolfgangMau, 2018-01-21:
I've played around with go only for a couple of weeks, but have tried serval gui approaches .. all of this was not really satisfying, since when it comes to third-party bindings, like Qt per example, then 
 the 'works on my machine'-effect comes into place again.
Some people like to fake a 'gui' with web-server and webviews (in my eyes very ugly)
I like Go because of the static linked binaries and the 'easy-to-use' cross-compiler , which makes it very handy for small cli-tools.
Unfortunately I didn't found a cool & easy to use xmodem library .. would have that integrated too, like in the chamtool.py

the chamtool actually didn't work with the revE-rebooted ... could probably be adapted
because it works with the revG - thus should be easy for a python-guy to make this work ...
I should have learned python, so maybe a good point to start with ... looks not too complex :)

WolfgangMau, 2018-01-21:
if someone grant me branch-permissions, I could push it ;-)

WolfgangMau, 2018-01-21:
No, I don't know that much about github ;-)

I don't really want to collect forks in this account - I used to create feature-branches, which got merged, once they are finished, or deleted if they never got finished.
Otherwise I would have my account full of forks with minor changes at first, and getting into a merge-hell later on.
I don't like the pm3-fork-hell either - you always have to clone another fork, just to have a single feature ... maybe I'm wrong, but thats sounds not that practical to me

bogiton, 2018-01-21:
Good job @WolfgangMau.
I was playing with chamtool and chamlog a few weeks ago. The main reason it wasn't working was the "MY" extension of the terminal commands that the rebooted firmware has. I think that after I added that, it was working ok (for the supported commands at least).
I will test your version later tonight.

WolfgangMau, 2018-01-21:
it was a little bit more than just the commands - as I mentioned above ... you would see after diffing ;-)

iceman1001, 2018-01-21:
yeah.. that xmodem sure does take space on the device.  In order to up/download,  doesn't MF_DETECTION skip xmodem? Would be nice just to get rid of it.

bogiton, 2018-01-21:
To be honest, I am not sure if Linux/OSX users would use a GUI instead of the command line tools.
Personally, I would go with a rebooted chamtool, since it is already there.

WolfgangMau, 2018-01-21:
even if it's just for me - I would like to have a GUI ... since it's more comftable

iceman1001, 2018-01-21:
... I saw others make a golang client for pm3 and the crosscompilation is amazing.   You get support on so many OS's outof the box.

bogiton, 2018-01-21:
Hmm, ok then we should give it a try :P

iceman1001, 2018-01-21:
Well.. First we make windows GUI that we like and we are happy with.

Then someone who likes **Go** can do a GUI based on the Windows version :)

WolfgangMau, 2018-01-21:
the static go ui (libui) is very limited - so no fancy shit at all :-D 
simple but working (mostly) ... more can I not achieve here with Golang

WolfgangMau, 2018-02-19:
[mostly working version for windows, linux & osx](https://github.com/WolfgangMau/chamgo-qt/releases)
due to my 'one man show' and missing knowledge about C, QT and Go ... the sourcode looks like crap - but it works ( at least for me on osx high sierra)
the mfke32v2 thingy works with a workaround, since I was not able to migrate the c-code to go-code and also I didn't get it to work which including the c-code (should be possible, but I didn't get it managed in a reasonable time) 
windows (windows7 32 bit) & linux (debian 9.3.0 64bit) was tested on virtualbox
osx was tested on a native macbook

iceman1001, 2018-02-19:
Nice!

iceman1001, 2021-07-02:
closing since no action

closed 2021-07-02 by iceman1001