iceman1001/ChameleonMini-rebooted #7: GUI source code

iceman1001, 2017-11-30:
I've asked the manufacturer about getting the source code for the GUI.  They said they will look into it.

It is a start.

gtpy, 2017-11-30:
Great. We’ll look forward.

iceman1001, 2017-12-04:
Some bad news, 
The student who wrote the GUI does not want it to be open-sourced...  

I guess I will have to write my own GUI....

gtpy, 2017-12-04:
Too bad. So we need to make our own. 
I'll take a closer look to it if I find some time between family and job.
December is really stressful :-)

iceman1001, 2017-12-04:
I know.  Time is always short when it comes to these things.

kevin2008-01, 2017-12-04:
i search for partial decompile it. see you later.

iceman1001, 2017-12-06:
Part success! 
I got the gui code for monitoring the device.

iceman1001, 2017-12-06:
https://pastebin.com/3Y6SGFi9

iceman1001, 2017-12-14:
I seem to forgotten, but someone sent me their own GUI.   **ChameleonMiniGui v0008_s_2.exe**
I managed find time to play with it now.  It has some interesting features.  
It would be nice having both the rebooted gui and this one.   There is potential here!

And all those minor issues with missing libs,  either VC++ (different languages)  or mscorlibs..
This needs to be adjusted so all pre-compiled software gets a nice package.  Much windows tools in this.

iceman1001, 2017-12-14:
And it was from @kevin2008-01   in another issue.   Perfect,  does this GUI come from the original Chameleon Mini repo or?!?

kevin2008-01, 2017-12-14:
yes sure, original. just do it iceman......

kevin2008-01, 2017-12-14:
i don't test with boomerang for decompil it. If i have time i test for you.

bogiton, 2017-12-17:
Is it possible to get any more source code for the GUI or it would be better to start building one from scratch? Maybe in another language too, like C#? What do you guys think?

iceman1001, 2017-12-17:
I suspect a brand new GUI based on C# would be the best.  
However getting inspiration from existing GUI's    Taking the best parts and drop the rest.

1. one advantage with this rebooted is the mfkey attack implemented.  So getting those attack inside it would be good.  
2. the current chameleon can do much on the deviceside.  Which is encouraged.
3. save / loading / comparing dumps.
4. creating / converting dumps between MCT/PM3/LIBNFC etc.

kevin2008-01, 2017-12-17:
==> Iceman, great !!!! What do you want....
I discuss with papy, and according for christmas, i receive one VIGIK VTI25M for testing the chameleon.... MF Detect.... Ready please.....
Please implement +1 / -1 UID, and for the future roll'up...
Sincerely.

iceman1001, 2017-12-17:
Vigik ?  hm,  chameleon has problems with faking a card against it.
Some of the fixes needed is,  respond with a different sak then inside s0-block0.

bogiton, 2017-12-19:
@iceman1001 I started working on the C# GUI we were discussing. Which is the best way to share this? PR?

bogiton, 2017-12-19:
[ChameleonMiniRebootedGUI-alpha_source.zip](https://github.com/iceman1001/ChameleonMini-rebooted/files/1572008/ChameleonMiniRebootedGUI-alpha_source.zip)

This was easier for me, hope you don't mind.

iceman1001, 2017-12-19:
The question is good.  Shall we have a new repo for it?

bogiton, 2017-12-19:
I suppose we could just include it under the /Software/ folder.

iceman1001, 2017-12-19:
well.. it depends on how big it will become?  no?

gtpy, 2017-12-19:
I think an own repo will be good, links can be added to each other. So we can maybe develop the GUI fore more purposes. What do you think?

-------- Original-Nachricht --------

iceman1001, 2017-12-19:
I do like the own repo,  with its own issues and wiki maybe..

iceman1001, 2017-12-19:
And its up:   https://github.com/iceman1001/ChameleonMini-rebootedGUI

I will close this issue,   and add that pastebin link in a new issue there..

closed 2017-12-19 by iceman1001

iceman1001, 2017-12-19:
So this is the gui dude,   http://35b0.de/2017/12/18/chameleonmini-gui/
Is this @bogiton  or @kevin2008-01  ?!?

bogiton, 2017-12-19:
Don't know what that is. I only uploaded it here. Is that for the rebooted version?

iceman1001, 2017-12-19:
No,  that was someone elses GUI..  
Your contribution is the foundation in the new repo,  
https://github.com/iceman1001/ChameleonMini-rebootedGUI

kevin2008-01, 2017-12-20:
this GUI no working with rebooted....

iceman1001, 2017-12-20:
and that would be because its not a finished GUI software.  I'm sorry if that wasn't clear.