iceman1001/ChameleonMini-rebooted #10: Installning instructions

iceman1001, 2017-12-14:
Looking at this repo,  trying the source code, it comes to me that we need some installation guidelines.

- Setting up a development environment
- compiling
- flashing

On different systems.

kevin2008-01, 2017-12-14:
yes, sure. will be patient...

ceres-c, 2017-12-16:
I would help you, if only I had the hardware...

iceman1001, 2017-12-16:
x-mas is coming?

gtpy, 2017-12-16:
I could help to write some instructions. Own a rev. G but it is not so different.

-------- Original-Nachricht --------

iceman1001, 2017-12-16:
@gtpy  perfect!

gtpy, 2017-12-16:
@iceman1001 How you want the documentation - in wiki or seperate file?

ceres-c, 2017-12-16:
@iceman1001 It is coming and it brought an XPS 13 to replace my broken Surface Pro. I don't know whether Santa has got any more money or not :')

gtpy, 2017-12-16:
santa wants to bring me a proxmark3 rdv2 but i think not this year :')
Wishes of my kids have more priority ;-)

iceman1001, 2017-12-16:
@gtpy  Go for the Wiki.  Easier for everyone to contribute.

gtpy, 2017-12-16:
Ok - I'll go for it in the evening.

ceres-c, 2017-12-16:
@gtpy Santa clearly has got his priorities straight. Being myself "the kid" in our family gives me privileges. ;-)
Going back on topic: I'll try to help as soon as I have the required hardware. Sorry guys but I can't help here :-/

iceman1001, 2017-12-16:
well..  @ceres-c   you can always start merging code from revG  :)

ceres-c, 2017-12-16:
I'll work on it during Christmas break, I hope :)

ceres-c, 2017-12-16:
I'might work on it during Christmas break, I hope :) 
Still, without a test device I can't tell whether it works or not

gtpy, 2017-12-17:
@iceman1001 
I don't own a Rev. E rebooted: which core of the hardware is formed on it? Rev. G has an Atmel ATXMega128A4U
Need it for some details in the Wiki.
Cheers

iceman1001, 2017-12-17:
Same as the RevE in the offical repo ;)    atxmega32a4u

gtpy, 2017-12-18:
..first instructions are added..
Cheers

iceman1001, 2017-12-18:
its a good start.

bogiton, 2017-12-18:
Although they don't qualify as installation instructions at all, I think many people will find it interesting if we included the supported commands in the Wiki.
When connected through a serial terminal, the supported commands for the rebooted device are the following:
VERSIONMY,CONFIGMY,UIDMY,READONLYMY,UPLOADMY,DOWNLOADMY,RESETMY,UPGRADEMY,MEMSIZEMY,UIDSIZEMY,BUTTONMY,SETTINGMY,CLEARMY,HELPMY,RSSIMY,DETECTIONMY

gtpy, 2017-12-18:
Right. We can add some pages with commands and configuration options

-------- Original-Nachricht --------

iceman1001, 2017-12-18:
Definitly!   Everything that makes working with this device easier,   or brings an understanding to it,  should be on the wiki.
Go for it. 

The whole support system around this tool seems so off.  Nothing is there. Its a void that needs to be fill!

gtpy, 2017-12-18:
@bogiton are the commands really like VERSIONMY?
I know it from Rev.G as VERSION?

iceman1001, 2017-12-18:
one of the strangeness and isolation of the code,  they changed the commands..

https://github.com/iceman1001/ChameleonMini-rebooted/blob/master/Firmware/Chameleon-Mini/Terminal/Commands.h

bogiton, 2017-12-18:
Yes, these are the commands that the device with the default firmware answers to. In order to be sure before providing them, I sniffed the traffic between the COM port and the GUI.

gtpy, 2017-12-18:
Yeah of course :D my fault - was looking in my local (wrong) repo.
Only one day to work, then I'll have holidays and maybe a little bit more time for projects.

kevin2008-01, 2017-12-18:
For me, if you want, i write wiki for the mf detect, right ? and i post many results and expériences... any time i test on many reader.... random uid... extract...

gtpy, 2017-12-18:
How to set up the development environment has started.

iceman1001, 2017-12-19:
We can get inspiration from the offical repo documentations.
https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html

icsom, 2018-01-12:
let me know if I can help on Linux or OSX
at the moment I can not even make the firmware :-(
but got a RevE Rebooted today

iceman1001, 2018-01-13:
Linux, I think it should be doable to down a dev-env suitable for compiling it .  The current firmware should compile.
You are most welcome to make thing better,  both on linux and OSX.

WolfgangMau, 2018-01-13:
yes, I managed it yesterday to compile with OSX and added the needed steps to the wiki (unfortunately under my other Name ;-))

iceman1001, 2018-01-13:
No worries @icsom  @WolfgangMau ,  same same but different.
Now we would need a OSX flashing wiki page.   @bogiton  did a nice bat-file, which takes the output epp/hex files and make the needed bin files for BOOT_LOADER... to work. 

If not,  try flashing to restore hex-file or the original rebooted image.  Let us know how you solve it on OSX :)   And yeah,  the command set is different,  look at wiki page in order to get the rebooted commands if you want to run a terminal window.

stek29, 2019-04-10:
Wiki is quite a mess, and I'm not sure where to post this.
avrdude flashes fine on macOS with avrispmkII.

```sh
# Set fuses (once)
avrdude -vv -p atxmega32a4 -P usb -c avrispmkII -U fuse0:w:0x00:m -U fuse1:w:0x00:m -U fuse2:w:0xbe:m -U fuse4:w:0xfe:m -U fuse5:w:0xec:m -U lock:w:0x3f:m
# Flash app and eeprom
avrdude -vv -p atxmega32a4 -P usb -c avrispmkII -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i
```

btw, building is also as simple as running "make" in Firmware folder.
Dependencies: avr-gcc, avr-binutils.
It's quite easy to install them with homebrew: `brew install osx-cross/avr/avr-gcc osx-cross/avr/avr-binutils avrdude`

shinhub, 2019-08-12:
Due to recent wiki changes, that describe working processes to compile and flash on Windows, OSX and GNU/Linux, could we consider it closed @iceman1001 ?

closed 2019-08-12 by iceman1001