iceman1001/ChameleonMini-rebooted #9: no device present

peseta3, 2017-12-14:
Hallo together,
I am a Newbee, I have the follow problem.
I make all I have to do with the getting started page

[http://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html](url)

And I get with win10 the Massage
   

> dfu-programmer: no device present.
>     There was an error with executing this command. Maybe your ChameleonMini is not in bootloader

 mode?

what can I do with this?

iceman1001, 2017-12-14:
As it recommends,  try to get your device in bootloader mode...

peseta3, 2017-12-14:
I did it

iceman1001, 2017-12-14:
You might be better off asking on the offical chameleon mini repo.  This repo is to work the firmware than offering support.

Good luck!

closed 2017-12-14 by iceman1001

peseta3, 2017-12-14:
hmm

peseta3, 2017-12-14:
can you give me the
offical chameleon mini repo

ceres-c, 2017-12-14:
It's the one where you already opened the issue https://github.com/emsec/ChameleonMini/issues/158
:)

peseta3, 2017-12-14:
ok
but they says that I hav to go to you.
because they have an other device

what can I now make?

peseta3, 2017-12-14:
they says I have to go to Iceman

ceres-c, 2017-12-14:
Yeah, I saw they told you to come here, but this is not yet a stable project. We're not working on fixing bugs at the moment, we're still trying to understand and fix the codebase.
You have to understand if you own hardware rev G or hardware rev E. Where did you buy it?

Also, at this moment I am not working on it due to real life issues.

peseta3, 2017-12-14:
I booght it from Lab401.com it is the RevE

peseta3, 2017-12-14:
ok real life has priority

ceres-c, 2017-12-14:
> I bought it from Lab401.com it is the RevE
So you better ask them.
Are you using linux?

peseta3, 2017-12-14:
I ask them but I didn`t get answere

I have both

Win10 and ubuntu 16

ceres-c, 2017-12-14:
Try to run a `dmesg -w` whthout the chameleon attached to the PC, then connect the device and post the output

peseta3, 2017-12-14:
-w it doesn`t work

peseta3, 2017-12-14:
sorry ubuntu 12.04

ceres-c, 2017-12-14:
Oh, ok. Then run `dmesg` without the chameleon and another `dmesg` 10 seconds after you connect the device

peseta3, 2017-12-14:
combining
[    0.655975] mtrr: type mismatch for b0000000,4000 old: write-back new: write-combining
[    0.655976] mtrr: type mismatch for b0000000,2000 old: write-back new: write-combining
[    0.655978] mtrr: type mismatch for b0000000,1000 old: write-back new: write-combining
[    0.656420] vesafb: framebuffer at 0xb0000000, mapped to 0xffffc90010780000, using 4160k, total 4160k
[    0.659113] Console: switching to colour frame buffer device 170x48
[    0.661653] fb0: VESA VGA frame buffer device
[    0.661670] intel_idle: does not run on family 6 model 61
[    0.661675] ipmi message handler version 39.2
[    0.662087] ACPI: Deprecated procfs I/F for AC is loaded, please retry with CONFIG_ACPI_PROCFS_POWER cleared
[    0.662736] ACPI: AC Adapter [ADP0] (off-line)
[    0.662817] input: Lid Switch as /devices/LNXSYSTM:00/device:00/PNP0C0D:00/input/input0
[    0.663137] ACPI: Lid Switch [LID0]
[    0.663175] input: Power Button as /devices/LNXSYSTM:00/device:00/PNP0C0C:00/input/input1
[    0.663179] ACPI: Power Button [PWRB]
[    0.663207] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
[    0.663209] ACPI: Power Button [PWRF]
[    0.664084] Monitor-Mwait will be used to enter C-1 state
[    0.664095] Monitor-Mwait will be used to enter C-2 state
[    0.664105] Monitor-Mwait will be used to enter C-3 state
[    0.664120] ACPI: acpi_idle registered with cpuidle
[    0.665040] GHES: HEST is not enabled!
[    0.665122] Serial: 8250/16550 driver, 32 ports, IRQ sharing enabled
[    0.668863] Linux agpgart interface v0.103
[    0.670266] brd: module loaded
[    0.670886] loop: module loaded
[    0.671220] libphy: Fixed MDIO Bus: probed
[    0.671311] tun: Universal TUN/TAP device driver, 1.6
[    0.671312] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[    0.671360] PPP generic driver version 2.4.2
[    0.671398] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.671402] ehci-pci: EHCI PCI platform driver
[    0.671529] ehci-pci 0000:00:1d.0: EHCI Host Controller
[    0.671534] ehci-pci 0000:00:1d.0: new USB bus registered, assigned bus number 1
[    0.671546] ehci-pci 0000:00:1d.0: debug port 2
[    0.675452] ehci-pci 0000:00:1d.0: cache line size of 64 is not supported
[    0.675477] ehci-pci 0000:00:1d.0: irq 23, io mem 0xc1219000
[    0.686542] ehci-pci 0000:00:1d.0: USB 2.0 started, EHCI 1.00
[    0.686589] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.686591] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.686593] usb usb1: Product: EHCI Host Controller
[    0.686594] usb usb1: Manufacturer: Linux 3.13.0-117-generic ehci_hcd
[    0.686596] usb usb1: SerialNumber: 0000:00:1d.0
[    0.686759] hub 1-0:1.0: USB hub found
[    0.686768] hub 1-0:1.0: 2 ports detected
[    0.686873] ehci-platform: EHCI generic platform driver
[    0.686880] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.686882] ohci-pci: OHCI PCI platform driver
[    0.686889] ohci-platform: OHCI generic platform driver
[    0.686895] uhci_hcd: USB Universal Host Controller Interface driver
[    0.686990] xhci_hcd 0000:00:14.0: can't derive routing for PCI INT A
[    0.686992] xhci_hcd 0000:00:14.0: PCI INT A: no GSI - using ISA IRQ 7
[    0.687019] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    0.687024] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[    0.688091] xhci_hcd 0000:00:14.0: cache line size of 64 is not supported
[    0.688112] xhci_hcd 0000:00:14.0: irq 61 for MSI/MSI-X
[    0.688165] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002
[    0.688167] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.688168] usb usb2: Product: xHCI Host Controller
[    0.688170] usb usb2: Manufacturer: Linux 3.13.0-117-generic xhci_hcd
[    0.688172] usb usb2: SerialNumber: 0000:00:14.0
[    0.688350] hub 2-0:1.0: USB hub found
[    0.688364] hub 2-0:1.0: 11 ports detected
[    0.692162] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    0.692168] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 3
[    0.692200] usb usb3: New USB device found, idVendor=1d6b, idProduct=0003
[    0.692202] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.692204] usb usb3: Product: xHCI Host Controller
[    0.692205] usb usb3: Manufacturer: Linux 3.13.0-117-generic xhci_hcd
[    0.692207] usb usb3: SerialNumber: 0000:00:14.0
[... 285 lines cut ...]
[  713.964754] wlan0: association with e0:28:6d:45:3b:d6 timed out
[  720.771224] wlan0: authenticate with e0:28:6d:45:3b:d6
[  720.790996] wlan0: send auth to e0:28:6d:45:3b:d6 (try 1/3)
[  720.793294] wlan0: authenticated
[  720.794306] wlan0: associate with e0:28:6d:45:3b:d6 (try 1/3)
[  720.799015] wlan0: RX AssocResp from e0:28:6d:45:3b:d6 (capab=0x431 status=0 aid=1)
[  720.799222] wlan0: associated
[  720.799264] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  720.799375] cfg80211: Calling CRDA for country: DE
[  720.805239] cfg80211: Regulatory domain changed to country: DE
[  720.805250] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[  720.805258] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm)
[  720.805264] cfg80211:   (5150000 KHz - 5250000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[  720.805269] cfg80211:   (5250000 KHz - 5350000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[... cut ...]
[  965.680476] usb 2-2: New USB device found, idVendor=03eb, idProduct=2044
[  965.680480] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[  965.680482] usb 2-2: Product: LUFA CDC Demo
[  965.680484] usb 2-2: Manufacturer: Dean Camera
[  965.680486] usb 2-2: SerialNumber: 438503937383FFFFB0FFB1006000
[  965.680641] usb 2-2: ep 0x82 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[  965.680992] cdc_acm 2-2:1.0: ttyACM0: USB ACM device
[ 1024.175513] usb 2-2: USB disconnect, device number 11

peseta3, 2017-12-14:
[    0.661653] fb0: VESA VGA frame buffer device
[    0.661670] intel_idle: does not run on family 6 model 61
[    0.661675] ipmi message handler version 39.2
[    0.662087] ACPI: Deprecated procfs I/F for AC is loaded, please retry with CONFIG_ACPI_PROCFS_POWER cleared
[    0.662736] ACPI: AC Adapter [ADP0] (off-line)
[    0.662817] input: Lid Switch as /devices/LNXSYSTM:00/device:00/PNP0C0D:00/input/input0
[    0.663137] ACPI: Lid Switch [LID0]
[    0.663175] input: Power Button as /devices/LNXSYSTM:00/device:00/PNP0C0C:00/input/input1
[    0.663179] ACPI: Power Button [PWRB]
[    0.663207] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input2
[    0.663209] ACPI: Power Button [PWRF]
[    0.664084] Monitor-Mwait will be used to enter C-1 state
[    0.664095] Monitor-Mwait will be used to enter C-2 state
[    0.664105] Monitor-Mwait will be used to enter C-3 state
[    0.664120] ACPI: acpi_idle registered with cpuidle
[    0.665040] GHES: HEST is not enabled!
[    0.665122] Serial: 8250/16550 driver, 32 ports, IRQ sharing enabled
[    0.668863] Linux agpgart interface v0.103
[    0.670266] brd: module loaded
[    0.670886] loop: module loaded
[    0.671220] libphy: Fixed MDIO Bus: probed
[    0.671311] tun: Universal TUN/TAP device driver, 1.6
[    0.671312] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[    0.671360] PPP generic driver version 2.4.2
[    0.671398] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.671402] ehci-pci: EHCI PCI platform driver
[    0.671529] ehci-pci 0000:00:1d.0: EHCI Host Controller
[    0.671534] ehci-pci 0000:00:1d.0: new USB bus registered, assigned bus number 1
[    0.671546] ehci-pci 0000:00:1d.0: debug port 2
[    0.675452] ehci-pci 0000:00:1d.0: cache line size of 64 is not supported
[    0.675477] ehci-pci 0000:00:1d.0: irq 23, io mem 0xc1219000
[    0.686542] ehci-pci 0000:00:1d.0: USB 2.0 started, EHCI 1.00
[    0.686589] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.686591] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.686593] usb usb1: Product: EHCI Host Controller
[    0.686594] usb usb1: Manufacturer: Linux 3.13.0-117-generic ehci_hcd
[    0.686596] usb usb1: SerialNumber: 0000:00:1d.0
[    0.686759] hub 1-0:1.0: USB hub found
[    0.686768] hub 1-0:1.0: 2 ports detected
[    0.686873] ehci-platform: EHCI generic platform driver
[    0.686880] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.686882] ohci-pci: OHCI PCI platform driver
[    0.686889] ohci-platform: OHCI generic platform driver
[    0.686895] uhci_hcd: USB Universal Host Controller Interface driver
[    0.686990] xhci_hcd 0000:00:14.0: can't derive routing for PCI INT A
[    0.686992] xhci_hcd 0000:00:14.0: PCI INT A: no GSI - using ISA IRQ 7
[    0.687019] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    0.687024] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 2
[    0.688091] xhci_hcd 0000:00:14.0: cache line size of 64 is not supported
[    0.688112] xhci_hcd 0000:00:14.0: irq 61 for MSI/MSI-X
[    0.688165] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002
[    0.688167] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.688168] usb usb2: Product: xHCI Host Controller
[    0.688170] usb usb2: Manufacturer: Linux 3.13.0-117-generic xhci_hcd
[    0.688172] usb usb2: SerialNumber: 0000:00:14.0
[    0.688350] hub 2-0:1.0: USB hub found
[    0.688364] hub 2-0:1.0: 11 ports detected
[    0.692162] xhci_hcd 0000:00:14.0: xHCI Host Controller
[    0.692168] xhci_hcd 0000:00:14.0: new USB bus registered, assigned bus number 3
[    0.692200] usb usb3: New USB device found, idVendor=1d6b, idProduct=0003
[    0.692202] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.692204] usb usb3: Product: xHCI Host Controller
[    0.692205] usb usb3: Manufacturer: Linux 3.13.0-117-generic xhci_hcd
[    0.692207] usb usb3: SerialNumber: 0000:00:14.0
[    0.692312] hub 3-0:1.0: USB hub found
[    0.692320] hub 3-0:1.0: 4 ports detected
[    0.702693] i8042: PNP: PS/2 Controller [PNP0303:PS2K,PNP0f13:PS2M] at 0x60,0x64 irq 1,12
[    0.725967] serio: i8042 KBD port at 0x60,0x64 irq 1
[    0.725973] serio: i8042 AUX port at 0x60,0x64 irq 12
[    0.726090] mousedev: PS/2 mouse device common for all mice
[... 287 lines cut ...]
[  720.799375] cfg80211: Calling CRDA for country: DE
[  720.805239] cfg80211: Regulatory domain changed to country: DE
[  720.805250] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[  720.805258] cfg80211:   (2400000 KHz - 2483500 KHz @ 40000 KHz), (N/A, 2000 mBm)
[  720.805264] cfg80211:   (5150000 KHz - 5250000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[  720.805269] cfg80211:   (5250000 KHz - 5350000 KHz @ 40000 KHz), (N/A, 2000 mBm)
[  720.805275] cfg80211:   (5470000 KHz - 5725000 KHz @ 40000 KHz), (N/A, 2698 mBm)
[  948.401265] usb 2-2: USB disconnect, device number 10
[  965.662025] usb 2-2: new full-speed USB device number 11 using xhci_hcd
[  965.680476] usb 2-2: New USB device found, idVendor=03eb, idProduct=2044
[  965.680480] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[  965.680482] usb 2-2: Product: LUFA CDC Demo
[  965.680484] usb 2-2: Manufacturer: Dean Camera
[  965.680486] usb 2-2: SerialNumber: 438503937383FFFFB0FFB1006000
[  965.680641] usb 2-2: ep 0x82 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[... cut ...]
[ 1050.940468] usb 2-2: new full-speed USB device number 12 using xhci_hcd
[ 1050.959406] usb 2-2: New USB device found, idVendor=03eb, idProduct=2044
[ 1050.959415] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[ 1050.959421] usb 2-2: Product: LUFA CDC Demo
[ 1050.959425] usb 2-2: Manufacturer: Dean Camera
[ 1050.959429] usb 2-2: SerialNumber: 438503937383FFFFB0FFB1006000
[ 1050.959693] usb 2-2: ep 0x82 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[ 1050.960167] cdc_acm 2-2:1.0: ttyACM0: USB ACM device

ceres-c, 2017-12-14:
It connects and is then disconnected. You could try and change the cable or it's an hardware issue, IMHO

peseta3, 2017-12-14:
I use the cable from the buying chameleon.
hmm ok I change it
what is IMHO

peseta3, 2017-12-14:
from win10 and an other cable it is the same problem

ceres-c, 2017-12-14:
> what is IMHO

In My Humble Opinion ;)

> from win10 and an other cable it is the same problem

Then it looks like an hardware fault. Is this device brand new?

peseta3, 2017-12-14:
hmmm
I find  a Chameleon RevE Rebooted Software
with
BOOT_LOADER_EXE.exe
Chameleon_EN.exe
myfile.bin
myfilee.bin

an I try the Chameleon_EN.exe   and click there to BootMode.
After that, I click the BOOT_LOADER_EXE.exe, And than he do something.
I have a picture from That.

peseta3, 2017-12-14:
it is a brand new from 2 Tays ago

peseta3, 2017-12-14:
ok 

thank you for your help
and your time

iceman1001, 2017-12-14:
Don't have to hardware.  It could be modemmanager on ubuntu messing up.
or on win10,  you need to install two drivers.    One for the normal operation,  the second for dfu-mode (bootloader)..   If you didn't,  the device will be not enumerated by your OS.

peseta3, 2017-12-14:
witch 2 drive I need for win 10?

peseta3, 2017-12-14:
If Have one for the bootloader modus

ATxmega32A4U

and

Normal Driver to connect it

Serielles USB-Device(COM8)

iceman1001, 2017-12-14:
The drivers was a bit hidden before.  So I added those two I got laying around in the Chameleon Mini folders.

/Drivers

https://github.com/iceman1001/ChameleonMini-rebooted/commit/45a05552d0048698d1d4f22c61cfc423a44f3254

peseta3, 2017-12-14:
but in the Driver folder isn`t a .inf ...........

iceman1001, 2017-12-14:
I rest my case.   Good luck!

peseta3, 2017-12-14:
I don`t understand!