iceman1001/ChameleonMini-rebooted #12: Maybe a way How to flash a new firmware on the Chameleon on a linux-based OS?

gtpy, 2017-12-23:
Hi

Maybe this is the way to flash on a linux based system:

1. Install DFU-Programmer (or be sure that the DFU-Programmer installation was successfull)

2. Activate Bootloader-mode on the ChameloenMini by holding down the button while powering on.
    (Black button)

3. Check/Confirm that the device is available 
    `lsusb`

4. Run the Firmware update / Program the device with dfu-programmer (maybe as root)
    `dfu-programmer atxmega32a4u erase`
    `dfu-programmer atxmega32a4u flash-eeprom Chameleon-Mini.eep`
    `dfu-programmer atxmega32a4u flash Chameleon-Mini.hex`
    `dfu-programmer atxmega32a4u reset`

I can’t test it because I don’t own a Rev. E Rebooted
If someone like to test / double check it won’t be great.

Maybe another way is available with avrdude but I actual have not enough time between Xmas.
Cheers

bogiton, 2017-12-23:
Nah, this doesn't work for the rebooted device.
In fact, these are the same commands that the windows batch script uses for flashing the original Chameleon-Mini. Which -again- don't work for the rebooted device.
Not even the suggested way of flashing under Linux (using avrdude) works for the rebooted Chameleon.

gtpy, 2017-12-23:
@bogiton ok. What’s the difference between both devices that it wouldn’t work?

bogiton, 2017-12-23:
Wish I knew. Well, apart from the fact that they are using a bit different MCU (atxmega32a4u instead of atxmega128a4u), as you may have noticed. Check issue #11, it is very relevant.

iceman1001, 2018-01-06:
Those commands should work,  since we use un-altered hex/eep when compiling in Atmel Studio 7...

shinhub, 2019-08-14:
@gtpy please read Wiki pages on flashing. This should be OK now.
@iceman1001 yet another close request.

closed 2019-08-14 by gtpy