iceman1001/ChameleonMini-rebooted #14: converting commands

iceman1001, 2017-12-30:
The current CMDMY convention feels out of date and a hinder somewhat with having a unified way for tools that work with Chameleon Mini RevG,  or this rebooted version. 
There are not **real** cause for this change of commands more than locking in the rebooted firmware from the offical one.   With putting the rebooted software open-software again, the reason falls away.

However, I do feel we should keep all the extras which makes the rebooted so special,  like the mfkeys implementation.   I can also see we incoorporate the hardnested attack also into the client in the long run. 

What you think?

bogiton, 2017-12-30:
I agree that the "MY" extension is unnecessary as well as kinda annoying. We should remove it.
I also agree that we should keep the rebooted extras, like the mfkey implementation, which by the way can be enabled by setting the "CONFIG_MF_DETECTION_SUPPORT" flag in the Makefile. Not sure if and how that works though.

iceman1001, 2017-12-30:
Well,  we can always add our own commands in order to have the extras.   As long as we follow the official revG commands behavior in order to keep inter-operability.

bogiton, 2017-12-30:
Indeed. Let's see if more people agree on that.

iceman1001, 2017-12-30:
I suppose with so few ppl active, this is kind of an easy decision ;)

ceres-c, 2017-12-30:
Agreed on both your proposals @iceman1001 
Also, sorry for my absence but I am really busy with exams in this period. I highly doubt my skills are sufficient for giving proper input to this project. I'll try to keep up with your updates and intervene when and if possilble

dunderhay, 2018-07-27:
+1 for same command set

iceman1001, 2019-03-10:
@bogiton   what do you say?  time to drop the **MY** stuff?  you up for it?

bogiton, 2019-03-10:
Ooh, yeah! Gladly! :)

iceman1001, 2019-03-10:
@bogiton   great!  don't forget to update the GUI to work with this version aswell

closed 2019-03-10 by iceman1001