iceman1001/ChameleonMini-rebooted #15: MFClassic_patch

orknist, 2018-01-04:
When I access a sector on emulator it's working as it should be.
But when I want to access more than one sector, communication typically fails after reading a few sectors.
It's seems like because of too slow response.

Could it be the same problem?
https://github.com/emsec/ChameleonMini/issues/83

Is the MFClassic_patch branch added into master?

iceman1001, 2018-01-04:
Yup, could very well be that.  if you compare the source code in that branch commented with this fork,  you see it doesn't use the modified crypto1 code.

iceman1001, 2018-01-06:
I have now tested the mfclassic patch.  It seems to work.  
@bogiton  will make a PR when he has finished his tests aswell.   I hope this will solve your problems

shinhub, 2019-08-14:
@orknist please try with new compiled firmware as of today, a lot have been reworked on Mifare.
@iceman1001 to be closed due to age, I guess.

closed 2019-08-14 by iceman1001