iceman1001/ChameleonMini-rebooted #16: eeprom / bootloader too big

iceman1001, 2018-01-04:
Trying to flash current code, I managed to get **Bootloader and code overlap.**
The eeprom / bin == 44bytes,  and below its expects 40bytes.     Not to mention that my device is now bricked...

```
C:\Chameleon RevE Rebooted Software\FlashTools>flash.bat
... (compressed for readability)
Creating the EEPROM binary...
Write done!
Creating the Flash binary...
Write done!

Flashing the files onto the "Chameleon-Mini Rev-E Rebooted"...
old_driver_bootloader
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
0%                            100%  Programming 0x40 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x400 bytes...
Bootloader and code overlap.
Use --suppress-bootloader-mem to ignore

If there are no errors above, flashing the firmware to your "Chameleon-Mini Rev-E Rebooted" should be finished now. Enjoy!
```

bogiton, 2018-01-04:
I had this issue too when was trying the different builds. The solution was to disable some of the features that I didn't need, like MF_CLASSIC_1K. Usually when you build the source, there is a warning telling that the code may overlap. But how did it brick the device? I flashed it over 30 times in the last week and it always came back with the "official" myfiles.

iceman1001, 2018-01-04:
It just doesn't want to flash the **offical mybin files** anymore.   Keeps my device in bootloader.

I belive the memory 32kb get used by static arrays in the firmware code for tag-memory.  If it was a more dynamic approach it would be better.

iceman1001, 2018-01-04:
bytearray used is called like this:   ApplicationProcess(CodecBuffer, DemodBitCount);
256 bytes only.  This would explain why 4K support doesnt work :)
However, it doesn't explain why bootloader grows..

```
#define CODEC_BUFFER_SIZE           256 /* Byte */
uint8_t CodecBuffer[CODEC_BUFFER_SIZE];
```

iceman1001, 2018-01-04:
Strange,  I can build/compile/flash with Atmel Studio 7 via the avmk-II devices,   no need for BOOT_LOADER_EXE..  So the encryption of mybin files seems to exists in order to prevent decompilation of the files itself.   Lets see if I can flash the old mybins files now.

bogiton, 2018-01-04:
That's a great find! I was using Atmel Studio for debugging but I've never tried flashing through that. If this works, it means that there is a pretty good chance for Linux users to be also able to flash using the AVR toolchain.

iceman1001, 2018-01-04:
[image]

**Setting of fuses**
This image shows how the fuses should be set, in order for automatic detection of battery power.
When battery is below 2.2v,  device can **not** start.
This ensures that read/write of external flash is stable.   Original chameleon mini doesn't need this since it has a power switch, it is not auto power on.

ceres-c, 2018-01-04:
It might be useful to post this in the wiki, don't you think?

Others might happen to brick their chimameleon...

iceman1001, 2018-01-04:
Sure,  feel free to create a page for using atmel studio :)

iceman1001, 2018-01-04:
well, I can flash it,  but the current build doesn't resurrect it..

ceres-c, 2018-01-04:
[Done](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Recovering-from-bootloader-and-code-overlap). Really, really badly done , but still done.

iceman1001, 2018-01-05:
:)  maybe a bit over the top, don't you think?

I haven't gotten the device functional again, still a learning process for me.

bogiton, 2018-01-05:
It is very weird that you can flash it without errors and still not having it working properly.
While I was testing the flashing of the firmware I was trying out the Atmel Flip software.
Using that, I was able to erase only the eeprom and program it with the .eep file. And after that, burn the firmware with Atmel Studio. Maybe this will help revive your device.

bogiton, 2018-01-05:
Haven't tried Atmel Studio for flashing yet. Probably Flip won't do much more, that is true.
I just came across this: [https://lab401.com/collections/hardware/products/avrisp-mkii?variant=5004374835231](https://lab401.com/collections/hardware/products/avrisp-mkii?variant=5004374835231)
Claims that it can unbrick the Chameleon RevE rebooted. But I guess you could also use a bus pirate or similar to do so.

iceman1001, 2018-01-05:
that is the device I have..  the **AVRISP-mkII**
Flashing is not an issue ;)   The file that I'm trying to flash seems to be an issue...

iceman1001, 2018-01-05:
Sloopy of me,   the orange flashing led should actually have told me something...
In the end I needed to unplug the battery in order for my AVRISP-mkII  to get a green light.
So now, the flashing wasn't any problem.   However my mini isnt working.

bogiton, 2018-01-05:
The following worked fine for my device:

- downloaded the master branch just now
- opened up the ITS_A_CARD.cproj file in Atmel Studio
- selected the "Release" configuration
- hit "Rebuild"
- copied the ITS_A_CARD.eep and ITS_A_CARD.hex from the Release directory to Chameleon-Mini.eep and Chameleon-Mini.hex accordingly
- run the flash.bat on them

FYI, here is the result of the build regarding memory usage:
```
Program Memory Usage 	:	21762 bytes   59,0 % Full
Data Memory Usage 		:	1861 bytes   45,4 % Full
EEPROM Memory Usage 	:	44 bytes   4,3 % Full
```

and here is the output from the flash.bat:
```
Flashing the files onto the "Chameleon-Mini Rev-E Rebooted"...
old_driver_bootloader
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
0%                            100%  Programming 0x40 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x400 bytes...
0%                            100%  Programming 0x5600 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
load_success!
```

iceman1001, 2018-01-06:
With @bogiton  's hex/eep files I got my mini working again.  It was the size of the files that messed it up.

I'm closing this one.

closed 2018-01-06 by iceman1001