iceman1001/ChameleonMini-rebooted #18: Mifare Emulation Problem

orknist, 2018-01-06:
I compiled and flashed latest master. Ultralight modes are working as good. 
But when I choose Mifare 1K or 4K mode, does not react at all. 
In addition if usb cable is not connected, always 8th led is swiched on. 
Do you have a prediction?

bogiton, 2018-01-06:
What do you mean "it does not react"?
I just tested both modes with my PM3. There seems to be some proximity issue, but *most* of the times it gets detected correctly.
Here is the commands I used to set the Chameleon up:
```
settingmy=1
100:OK
configmy?
101:OK WITH TEXT
MF_CLASSIC_1K
uidmy?
101:OK WITH TEXT
3146E4F8
settingmy=2
100:OK
configmy?
101:OK WITH TEXT
MF_CLASSIC_4K
uidmy?
101:OK WITH TEXT
2DDF0A06
```

And here is the output from the PM3:
```
pm3 --> hf 14a info
 UID : 31 46 E4 F8
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1
proprietary non iso14443-4 card found, RATS not supported
Answers to magic commands (GEN 1a): YES
Prng detection: WEAK

pm3 --> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:FF FF FF FF FF FF
#db# READ BLOCK FINISHED
isOk:01 data:31 46 E4 F8 6B 00 00 03 03 FF FF FF FF FF FF FF

pm3 --> hf 14a info
 UID : 2D DF 0A 06
ATQA : 00 02
 SAK : 18 [2]
TYPE : NXP MIFARE Classic 4k | Plus 4k SL1 | 4k Ev1
proprietary non iso14443-4 card found, RATS not supported
Answers to magic commands (GEN 1a): YES
Prng detection: WEAK

pm3 --> hf mf rdbl 0 A FFFFFFFFFFFF
--block no:0, key type:A, key:FF FF FF FF FF FF
#db# READ BLOCK FINISHED
isOk:01 data:2D DF 0A 06 FE FF FF FF FF FF FF FF FF FF FF FF
```

What reader do you use to test it?

About the LED problem, indeed there is an issue there. When powered by battery, you cannot switch slots. So, it stays always on the last selected slot from when the usb cable was connected. My guess is that the cycle settings function (SWITCHCARD) is not storing the selected slot when on battery. On the other hand, other button modes work normally. For example UID_RIGHT_DECREMENT.

iceman1001, 2018-01-06:
I have merge the crypto1 fixes from orignal revG into this,  however I can't test it...
@bogiton   Can I mail you?  or are you on signal?

bogiton, 2018-01-06:
Not using signal. Either e-mail (I sent you an e-mail a few minutes ago) or IRC :)

iceman1001, 2018-01-06:
cool,  email answered :)   otherwise  http://webchat.freenode.net/?channels=#proxmark3

orknist, 2018-01-06:
I have Proxmark3 RDV2, Oneplus 3 smartphone, ACR122U and 2 more type smartcard readers.
Everything looks good on PM3, but when I use others, it's acting like no card.
I mean that only working with PM3.

Chameleon's info;

```
SETTINGMY=0
100:OK
CONFIGMY?
101:OK WITH TEXT
MF_CLASSIC_1K
UIDMY?
101:OK WITH TEXT
00000001
SETTINGMY=1
100:OK
CONFIGMY?
101:OK WITH TEXT
MF_CLASSIC_4K
UIDMY?
101:OK WITH TEXT
00000002
```

PM3's output;

```
proxmark3> hf 14a reader
 UID : 00 00 00 01           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
Answers to chinese magic backdoor commands: YES  

proxmark3> hf mf rdsc 0 A FFFFFFFFFFFF
--sector no:0 key type:A key:ff ff ff ff ff ff            
#db# READ SECTOR FINISHED                 
isOk:01          
data   : 00 00 00 01 01 08 04 00 62 63 64 65 66 67 68 69           
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
trailer: ff ff ff ff ff ff ff 07 80 69 ff ff ff ff ff ff 

proxmark3> hf 14a reader
 UID : 00 00 00 02           
ATQA : 00 02          
 SAK : 18 [2]          
TYPE : NXP MIFARE Classic 4k | Plus 4k SL1          
proprietary non iso14443-4 card found, RATS not supported          
Answers to chinese magic backdoor commands: YES 

proxmark3> hf mf rdsc 0 A FFFFFFFFFFFF
--sector no:0 key type:A key:ff ff ff ff ff ff            
#db# READ SECTOR FINISHED                 
isOk:01          
data   : 00 00 00 02 02 08 04 00 62 63 64 65 66 67 68 69           
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
data   : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
trailer: ff ff ff ff ff ff ff 07 80 69 ff ff ff ff ff ff

```

iceman1001, 2018-01-06:
That could be because pm3 is more acceptable for timing etc in the protocol.
However,  there are fixes from RevG offical mini,  which I have merged,  I sent them to @bogiton  for him to test out.   If they work, we can merge it.

The magic commands is also a side-issue.   mini shouldn't answer to magic commands by default...
some readers has countermeasures which detects magic cards..

bogiton, 2018-01-06:
There is a flag to enable the magic commands feature while compiling. It is called "SUPPORT_MF_CLASSIC_MAGIC_MODE". Maybe we should remove it by default.
I will try to test the crypto1 fix today and let you guys know.

iceman1001, 2018-01-06:
in the source you got,  the MF_detect mode is "merged" into  CLASSIC 1K..  and removed.
I don't think we need a separate mode for it.  It should save some space too.  Since we don't have much to play with here.  There is the "encryption" of the collected data,  totally unneeded,  but if we remove it the existing GUI fails..   That would force ppl to use our GUI project here...

For proper CLASSIC 1K/4K emulation,  you need to compile without the **SUPPORT_MF_CLASSIC_MAGIC_MODE** flag in the Makefile.

kgamecarter, 2018-03-30:
I hava same problem.
Mifare 1K emulation work fine for PM3 reading.
But Android NFC and Door lock does not react.

kgamecarter, 2018-04-18:
I bought a AVR MK2 for flashing latest master.
And modified fuse register.
Now, All work fine.

iceman1001, 2018-04-18:
@kgamecarter   Nice!

iceman1001, 2018-04-18:
@orknist   have you also tested the lastest source ?

RenwickCustomer, 2018-06-22:
@kgamecarter I'm having the same issue, are you able to expand on the exact steps you took to resolve the issue please? Would be much appreciated!

shinhub, 2019-08-14:
@orknist , @RenwickCustomer please try with new compiled firmware as of today, a lot have been reworked on Mifare. Give a shot at wiki page to program with AVRISP or through USB if need be. @iceman1001 this one is more than 1 year old as well, if you would...

closed 2019-08-14 by iceman1001