iceman1001/ChameleonMini-rebooted #19: draft for release v1.0

iceman1001, 2018-01-07:
I guess current status should be a release,  it compiles and works with old GUI.

https://github.com/iceman1001/ChameleonMini-rebooted/releases/tag/untagged-09ff734155892b121c57

Tell me if its ok or not,  and what should change if not.

kevin2008-01, 2018-01-09:
Hi, 404 error

iceman1001, 2018-01-09:
hm,  can you see the draft on  /code/releases  ??

bogiton, 2018-01-09:
I can't access it either. Nothing under releases/tags too.

iceman1001, 2018-01-09:
ok.  so I need to add you as admin...

iceman1001, 2018-01-19:
So,  @bogiton   you can see the draft.  
Question,   where are we in this process?   What more do we need to fix before we can make the first release?

bogiton, 2018-01-19:
I just did some last minute testing and I think we are ready to make the first release!
RSSI still does not work, but I don't think that is very important at the moment.
As soon as we make the first working release, we can focus on the optimization of the mifare classic emulation.

iceman1001, 2018-01-19:
Since you fixed most things,  would you mind editing the draft-message,  in order to describe this first release the best?

bogiton, 2018-01-19:
I'm not considered very good in describing, but I gave it a try. Do you think we should also refer to the supported configurations/modes?

iceman1001, 2018-01-19:
Execellent
and yes,  I do think we should write that in order to spread information and knowledge.

iceman1001, 2018-01-19:
I removed this section since it is more for our roadmap here,  than for a release note.

_Next releases would probably deviate from the old firmware and try to get closer to the RevG functionality and features, without losing though the usability of the rebooted device._

bogiton, 2018-01-19:
Indeed, I think it's better now.

iceman1001, 2018-01-19:
and its live!

https://github.com/iceman1001/ChameleonMini-rebooted/releases/tag/v1.0

Great work!

closed 2018-01-19 by iceman1001

bogiton, 2018-01-19:
Cool :)