iceman1001/ChameleonMini-rebooted #21: Chameleon mini Chinese version

nurzhannn, 2018-01-15:
Chameleon mini Chinese version will not work with the original firmware because the Chinese version this not original microcontroller atxmega128a4u,replaced by atxmega32a4u

<<< https://github.com/iceman1001/ChameleonMini-rebooted >>> the Chinese gave the distorted firmware,computer does not see Chameleon-mini
need something to fix it in the firmware
someone have a working firmware ?

WolfgangMau, 2018-01-15:
[ChameleonMiniRDV2.0_ATxmega32A4U](https://github.com/iceman1001/ChameleonMini-rebooted/blob/master/Firmware/Compiled/ChameleonMiniRDV2.0_ATxmega32A4U.hex)
this one works at least partially
(no Mifare Ultralight, Random-ID and possibly no 'Reader' - at least on my side)
look at
[issue #11](https://github.com/iceman1001/ChameleonMini-rebooted/issues/11)

nurzhannn, 2018-01-15:
Thank you,what software are you using and driver ?

WolfgangMau, 2018-01-15:
can only tell about  hi I got it flashed on windows7

Drivers:
[LUFA-Driver](https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Drivers/LUFA%20Driver) for CDC Mode (normal mode)

DFU-Driver for uploading (bootloader mode)
on Windows7 I installed the driver from within [this zip](https://github.com/iceman1001/ChameleonMini-rebooted/blob/master/Software/Win32/dfu-programmer-win-0.7.2.zip) 

Software:
normal mode (Configuration): [Chameleon%20Mini%20RevE%20Rebooted_en](https://github.com/iceman1001/ChameleonMini-rebooted/blob/master/Software/Win32/Chameleon%20Mini%20RevE%20Rebooted_en.exe) 

in order to flash I needed to copy the [BOOT_LOADER_EXE](https://github.com/iceman1001/ChameleonMini-rebooted/blob/master/Software/Win32/BOOT_LOADER_EXE.exe) next to the [FlashTools](https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Software/FlashTools) and find me a working msvcr120.dll ... i guess thats all, but maybe I forgot something ;-)

may all be different on other windows-versions
I'm not a fan of windows, so not much experience here 
but I did not got it working on OSX for now

nurzhannn, 2018-01-15:
https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Drivers/LUFA%20Driver
 how did you install ?
I only have a microcontroller driver installed

bogiton, 2018-01-15:
https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing---windows
This info doesn't help?

iceman1001, 2018-01-15:
To be correct,  you need to find the msvcr12 compiled with debug info.  Hence the ending **d** in the filename  **msvcr120d.dll**.     It is usually found with a visual studio download.

I added one to the folder:
https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Software/FlashTools

The BOOT_LOADER_EXE should also been there when running it.

The two existing "reset" firmware files (under Compiled folder)

1. ChameleonMiniRDV2.0_ATxmega32A4U     (resets to basic function)
2. ITS_A_CARD.hex    (restore back original chameleon mini rebooted function)

**Recovery process:** 
If you screwed up your device,  flash 1,  then flash 2 with BOOT_LOADER_EXE.
if you not screwed up your deivce,   flash 2 with BOOT_LOADER_EXE.

WolfgangMau, 2018-01-15:
my original chameleon was able to emulate ultralight, was able to create random mf uids and was able to act as reader ... but after flashing the both files it can . only emulate mf & mfu ..
so it's far away from 'restore back original chameleon mini rebooted function'

so what did I wrong?
I would like to have the 'original chameleon mini rebooted function'

bogiton, 2018-01-15:
As far as I remember there was no reader configuration in my original firmware.
In any case, the official stock binaries can be found here: https://github.com/iceman1001/ChameleonMini-rebooted/files/1573135/bootloader-files.zip
which you can flash using the https://github.com/iceman1001/ChameleonMini-rebooted/raw/master/Software/Win32/BOOT_LOADER_EXE.exe

iceman1001, 2018-01-15:
hen you run **VERSIONMY?**  in a terminal session,   it says?

If you could act as READER,  then you must have used Chameleon Mini RevG firmware.  Which device do you have?  

Hold on,  how did you to that "files/"  link?!?  Which do I see those zip-files?!?  Is that some GitHub woodoo I haven't seen yet?

bogiton, 2018-01-15:
hahaha, it is from a comment of mine in issue 11 :P
https://github.com/iceman1001/ChameleonMini-rebooted/issues/11#issuecomment-352876568
Actually these are the files that were provided by the shop I got my device from (along with the BOOT_LOADER_EXE and the original GUI).

iceman1001, 2018-01-15:
Aha,  well,  we have **two bin files** which should be original.  
Then we have a few **ITS_A_CARD.hex** which should be original aswell..
at last we have  **ChameleonMiniRDV2.0_ATxmega32A4U**  which is to put it in basic bootload-mode.

So @WolfgangMau   I believe you should be able to get to square one again.

WolfgangMau, 2018-01-15:
@iceman1001 
with
ChameleonMiniRDV2.0_ATxmega32A4U
ITS_A_CARD.hex

VERSIONMY?
101:OK WITH TEXT
Chameleon-new-1.0

with your files
VERSIONMY?
101:OK WITH TEXT
Chameleon-new-1.0

but .... mfu works again - YEAH!

your files have different size than my generated myfile.bin and myfileee.bin
your myfile.bin  -> 23KB . (mine -> 23KB)
your myfileee.bin -> 23B (mine -> 4KB)

iceman1001, 2018-01-15:
firmware **myfile.bin** should be less than 32kb (otherwise it will not fit the device)
eeprom file  **myfilee.bin** should only be around 40bytes.

iceman1001, 2018-01-15:
I would say your **myfilee.bin** was causing some troubles.
You can bin-diff them

WolfgangMau, 2018-01-15:
thx very much for the files ... I'm working in Finland this week ... but I carried the pm3 and the cmr with me because I was hoping that I get that fixed somehow over this week ... finally it worked out just the second day - very big thanks to you guys @iceman1001  & @bogiton

iceman1001, 2018-01-15:
Well, we are happy aswell.   At least we have the option to fiddle with the firmware and make a better GUI now.  We are all learning how to play with this device. 

How is it going for @nurzhannn  ?   Have you also got your mini up again?

nurzhannn, 2018-01-16:
ChameleonMiniRDV2.0_ATxmega32A4U (resets to basic function)
ITS_A_CARD.hex (restore back original chameleon mini rebooted function)

flash two firmwares and read microcontroller:

[1.txt](https://github.com/iceman1001/ChameleonMini-rebooted/files/1633496/1.txt)

computer does not see nothing 
;*****************************************************************************
then checked these two files:
[myfile.bin.txt](https://github.com/iceman1001/ChameleonMini-rebooted/files/1633544/myfile.bin.txt)
[myfilee.bin.txt](https://github.com/iceman1001/ChameleonMini-rebooted/files/1633547/myfilee.bin.txt)

not working

bogiton, 2018-01-16:
What is with the ".txt" extension? The myfile and myfilee files should have the ".bin" extension only.
Are you sure your device is in bootloader mode?
What is the output of the BOOT_LOADER_EXE.exe when you run it?
There is a different procedure for the hex/eep files (that are created from compiling the code) and the pre-compiled .bin files.
For the latter, I believe, that the guide [Flashing windows](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing---windows) is very thorough and should suffice.
As for the hex/eep files there is the flash.bat script (as well as the rest of the needed files) located in [FlashTools](https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Software/FlashTools).

nurzhannn, 2018-01-16:
file extension "bin" not load on the forum
all that was inside file extension "bin" copied in the file extension "txt"
;*****************************************************************************************
First a microcontroller memory was erased and next i download firmware:
1.ChameleonMiniRDV2.0_ATxmega32A4U.hex
bootloader "ATXMEGA32a4U"  is was installed:
https://github.com/iceman1001/ChameleonMini-rebooted/tree/master/Drivers/DFU%20Driver
computer identified "atxmega32a4u".

Next I did everything according to the instructions: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing---windows
in "DEVICE MANAGER" There is nothing,zero reaction

bogiton, 2018-01-16:
Let's try this again.
Hold down the black button and then connect your device via USB in a windows PC. This should put the Chameleon in bootloader mode and you should be able to see it as "atxmega32a4u" in the device manager.
Next, put the following files in the same directory:
* myfile.bin
* myfilee.bin
* BOOT_LOADER_EXE.exe

And then run the BOOT_LOADER_EXE.exe.
What is the output of the cmd window?

nurzhannn, 2018-01-16:
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
0%                            100%  Programming 0x20 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x400 bytes...
0%                            100%  Programming 0x5B00 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
load_success!

iceman1001, 2018-01-16:
prefect!    Time to close this issue

nurzhannn, 2018-01-16:
in "DEVICE MANAGER" There is nothing,zero reaction when I connect "Chameleon-mini"

iceman1001, 2018-01-16:
have you installed the chameleon mini usb drivers?  Are there a new usb-device assigned to a comport?

nurzhannn, 2018-01-16:
how installed the chameleon mini usb drivers if computer not does see "chameleon-mini" ?

iceman1001, 2018-01-16:
read my previous answer.

nurzhannn, 2018-01-16:
new usb-device,no nothing

iceman1001, 2018-01-16:
unplug,  plug in,   no new usb device when looking in DeviceManager?

nurzhannn, 2018-01-16:
no

iceman1001, 2018-01-16:
I can't help you. I have no idea why your computer doesn't identify it.

nurzhannn, 2018-01-16:
Checking memory from 0x0 to 0x6FFF...  Empty.
0%                            100%  Programming 0x60 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x400 bytes...
0%                            100%  Programming 0x100 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
load_success!

Programming 0x60 bytes instead Programming 0x20 bytes
Programming 0x100 bytes instead Programming 0x5B00 bytes

nurzhannn, 2018-01-16:
"myfile.bin" and "myfilee.bin" size 1 kb for both

iceman1001, 2018-01-16:
Which bin-files are you using?  That is wrong sizes.
Did you use:   https://github.com/iceman1001/ChameleonMini-rebooted/files/1573135/bootloader-files.zip

Put those 2 bin files inside same folder as BOOT_LOADER_EXE.exe file.   
run BOOT_LOADER_EXE.exe

What is the output for that?

jackkleeman, 2018-01-16:
I can't get BOOT_LOADER_EXE.exe to work as dfu-programmer doesn't work on my pc, only on my mac. Can you tell me the equivalent dfu-programmer instructions? I try erase but it tried to erase much more than to 0x6FFF, and the erase fails at 0x7FF1

jackkleeman, 2018-01-16:
I have same issue btw, unrecognised except in bootloader

nurzhannn, 2018-01-16:
Thank you for your patience, everything is working!

iceman1001, 2018-01-16:
@nurzhannn   great!
@jackkleeman  what is the output from your BOOT_LOADER_EXE ?  (with correct bin-files in the same folder)

jackkleeman, 2018-01-16:
old_driver_bootloader
dfu-old-driver: no device present.

bogiton, 2018-01-16:
not in bootloader mode then.

jackkleeman, 2018-01-16:
It is definitely in bootloader mode and it works on my mac

iceman1001, 2018-01-16:
indeed,  have you put it in bootloader_mode?  
Like the instructions on Wiki https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing---windows states?

jackkleeman, 2018-01-16:
yes, and device manager sees it correctly

iceman1001, 2018-01-16:
You see **Atmel Usb Devices ATxmega32A4U** in device manager?

jackkleeman, 2018-01-16:
I see ATxmega32A4U and in properties the device type is Atmel USB Devices

iceman1001, 2018-01-16:
The dfu-programmer doesn't work.  

You must use BOOT_LOADER_EXE or AVR-mkII device to flash. 
which OS are you on?  Something is off here.  If the device is identified as ATxmega32A4U the BOOT_LOADER_EXE will find it...  
Possible,  that the device is not in bootloader mode,  you may have installed wrong drivers,  since you need two of them..

jackkleeman, 2018-01-16:
I am trying this on PC. Dfu-programmer works on mac so it is definitely in bootloader. I installed the dfu driver in this repo. Do I need any other driver for talking to bootloader?

iceman1001, 2018-01-16:
dfu-programmer works on mac, ok, what is the output from that one?  Im curious.

You need two drivers on PC,  so the device gets identifed correct in the two modes.
I have no idea it is on OSX.

jackkleeman, 2018-01-16:
I can't get recognised in LUFA mode right now.
`➜  Compiled git:(master) ✗ dfu-programmer atxmega32a4u erase            
Checking memory from 0x0 to 0x7FFF...  Not blank at 0x7FF1.
Erasing flash...  Success
Checking memory from 0x0 to 0x7FFF...  Not blank at 0x7FF1.`

iceman1001, 2018-01-16:
that is only the **erase** command.  Didn't you run the rest?

jackkleeman, 2018-01-16:
Yes. 
`➜  Compiled git:(master) ✗ sudo dfu-programmer atxmega32a4u flash ITS_A_CARD.hex
Checking memory from 0x0 to 0x58FF...  Empty.
0%                            100%  Programming 0x5900 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x8000 bytes...
[ X  ERROR
Memory read error, use debug for more info.
FAIL
Memory did not validate. Did you erase?`

After this the lights can come on in LUFA mode but its not recognised by mac or pc and the button doesnt move the light

iceman1001, 2018-01-16:
that doesn't look like your flash worked.
Did you try the ITS_A_CARD.hex that comes with this repo?

jackkleeman, 2018-01-16:
Yup

iceman1001, 2018-01-16:
What is the output from dfu-programer with the original hex?

jackkleeman, 2018-01-16:
That was the original hex. I was saying I also tried my compiled one.

bogiton, 2018-01-16:
dfu-programmer shows the exact same (erroneous) output in windows, where it also does not work.
As iceman mentioned earlier, the only two methods to flash the rebooted device are either using BOOT_LOADER_EXE in windows (with the .bin files) or the AVR-mkII device.

jackkleeman, 2018-01-16:
Ok I will try another computer. Thanks

jackkleeman, 2018-01-17:
I have the same issue on another computer. No worries as my AVR-mkII is en route. However I think this project really needs another solution! Do we have a way to reverse engineer the bootloader exe?

iceman1001, 2018-01-17:
Feel free to figure out what the BOOT_LOADER_EXE.exe / createbin.exe does.  I'm guessing it has a old dfu-programmer inside together with the aes encryption.  There is more info in the other issues or wiki.

Getting flash working for other than windows users would be great.

Skydev0h, 2018-05-24:
BOOT_LOADER_EXEC seems to (at least) use some library like https://github.com/dfu-programmer/dfu-programmer judging from some stuff in decompiled code:

[image]

[image]

[image]

That stuff looks VERY much like the following file:
https://github.com/dfu-programmer/dfu-programmer/blob/master/src/commands.c

And the main itself looks very much like the main of that dfu-programmer:
[image]

Createfile does not seem to use AES encryption, but it does much juggling and xoring with data (may be some standard algorithm or may be homebrew one):

[image]

[image]

bogiton, 2018-05-25:
Indeed, the BOOT_LOADER_EXE has bundled a possibly modified dfu-programmer. All the output text of the command window seem to match the dfu-programmer. Maybe to also set the fuses through that?

[10 more replies]

closed 2019-08-14 by iceman1001