iceman1001/ChameleonMini-rebooted #22: Broken for 4k in several ways

jackkleeman, 2018-01-16:
Hi I believe this firmware does not account for the different sector sizes, and the higher sector number of the 4k. See for example the code here:
https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Application/MifareClassic.c#L617
Which has different offsets for keys.

Furthermore, the MEM_SECTOR_ADDR_MASK is broken for blocks above 64d; the mask maps these blocks back to sector 0. 0xFC would work, although not for blocks above 128 as these sectors are bigger.

bogiton, 2018-01-16:
Yeah, there are a couple of issues in the MF Classic emulation in the current build.
The plan is to get a mostly working build that is close to the stock firmware (and make a release at that point) and afterwards apply the MF Classic patches (coming from the official RevG version) that should solve most problems.

iceman1001, 2018-01-16:
@jackkleeman   feel free to help out!

jackkleeman, 2018-01-16:
Hi guys, I have managed to brick it now so I will wait for my avrisp and then help out.

iceman1001, 2018-01-16:
Don't count it out yet.  Use the solutions mentioned in other thread and you should be able to **unbrick it**  :)

jackkleeman, 2018-01-16:
I can't even get it recognised as bootloader now. I could never get it recognised as bootloader on windows as well, so I was flashing using Mac and dfu programmer. I want to be able to develop easily so I need an avrisp anyways

nurzhannn, 2018-01-18:
"chameleon-mini" he can sniffing acr122u with "mifare_classic" between them ?

bogiton, 2018-01-18:
@jackkleeman there is a program called Zadig (http://zadig.akeo.ie/) which enables you to replace installed drivers very easily. When in bootloader mode (black button pressed etc.) launch Zadig, Options->List all devices, select the Atxmega32a4u from the list and replace its driver with libusb-win32. This may fix your device being recognized by BOOT_LOADER_EXE.

@nurzhannn as far as I recall, the RevE rebooted firmware doesn't have support for reader and sniffing capabilities. RevG device supports it. I guess that we can add it to our firmware some time in the future though.

jackkleeman, 2018-01-18:
@bogiton I actually already tried this but I can try again

iceman1001, 2018-01-18:
Hard to tell if the device is in bootloader mode if the drivers are all mixed up.

Regarding sniffing?  We have 32kb to play with,  that is very little.  I don't think sniffing will be inside the major firmware,  (the logging functionality)  , but in a custom firmware where you can remove other stuff to fit it in..   I have no need for it,  since I have proxmark3 for the heavylifting stuff. 

 @bogiton  and I talked about why you would need sniffing, and in the end its because you want to get the unknown key or password used.  For Ultralight Ev1 / NTAG its a password in clear you want.
For Mifare its the keys..   
We have mfkey to get mifare keys out,  when enables you to run darkside/nested/hardnested attacks in order to collect all keys. 
We just need to save the clear text password and a way to extract it for Ultralight/NTAG and that functionality doesn't take much space.  
With that you have basically what you would need to extract all data from a tag that you have.

Act as a reader,..  it not quite the purpose of a Mini, but sure,   if you want to use it as a reader its doable already with RevG. 

I think  ISO15693 support  ISO14443b  and other protocols are more of value for this project.  After all bug fixes of course

shinhub, 2019-08-07:
I confirm there are still bad things happening with 4K (at least, I had not have enough real-case 1K tests to tell for 1K), even after 0b5cc0a commit (McEloff/fix_4k_tags) - bad like heavy memory corruption.

I can reproduce issue starting at the first firmware I have where you can choose a function for "Button Long". The first stock firmware from GUI "Load Default" does not have the same issue.
I noticed it first because various readers did not manage to read a multi-purpose 4K tag dump I have (Chameleon does not blink at all, readers does not react of seems to fail).
The same 4K. dump however perfectly work when I clone it to a real Chinese or direct write physical tag.

Here is how to check on the issue easily:
- flash latest firmware (or previous one, but not first stock default one),
- set first slot to 4K, upload a 4K dump,
- for testing purpose, set 2nd slot to 1K, upload a dump (preferably one where you can easily identify bytes patterns, or even ASCII readable patterns),
- others slots can be set or cleared, it does not matter,
- have the Chameleon 1st slot read by one real-life reader which would try to access some blocks ("Read" mode on MCT for Android device would do, if you do not have a real-life reader),
- plug the Chameleon back to the GUI, download the 1st slot dump back, and compare it to the original you uploaded (with GUI "Dump management" tab for instance).

You will be able to notice that (somewhere near 1K for me), the 4K dump you download back is totally messed with, compared to your original 4K dump, and notably that parts of the 2nd slot 1K dump are being mixed within the 4K dump you downloaded back (you will be able to spot you 1K-dump easily identifiable bytes patterns, or ASCII readable patterns, within the 4k dump that has been downloaded back from Chameleon).
This seems like a memory issue to me.

I will try to run more test cases in order to check if I can trigger the same mess with 1k dumps only.
Previous real-life 1k tests I did worked perfectly with Chameleon, whichever firmware, but this might be due to the fact I only tested "UID only" applications without knowing.

shinhub, 2019-08-14:
@jackkleeman please try with new compiled firmware as of today, a lot have been reworked on Mifare, notably 4k. Try your chance with Wiki rescue kit pages to restore a proper DFU if need be.
@iceman1001 could probably be closed as last comment from user is 1 year and half now.

closed 2019-08-14 by iceman1001