iceman1001/ChameleonMini-rebooted #23: draft for release v1.1

iceman1001, 2018-01-19:
With this draft I open up for discussions over which functionality the next release should have.

Current firmware is workig with existing GUI.  As of this release we are free to deviate from old legacy and bring new/fixs  into action!

-  merge **MF_DETECTION** functionality into **MIFARE_CLASSIC_nK** ,  partly to save space and since it really doesn't need its own function.
- 
- **MF_CLASSIC 1K**  should not react to magic command default.  It should be an option.
- **MF_CLASSIC 1K**  SAK 0x08  - 0x88  ,  optional to set SAK ,  not to be read for Sector0-Block0.
- **MF_CLASSIC 1K**  crypto1 enhancements.
- tag slots,  only allow two slots to have 4K support,  in order to save memory

kevin2008-01, 2018-01-20:
waouuu! fine. thank you iceman1001

iceman1001, 2019-02-22:
@bogiton 
So what do you think?

bogiton, 2019-02-22:
Should we also remove the "MY" extension for this release?
Are there 2 slots that support 4K? I thought it was only the first one.

iceman1001, 2019-02-22:
Should be one,   but all this extra memory we now have...  Question is if we should be able to do 4k everywhere..
4*8kb in flash mem
and the active slot is downloaded to ram ?
..
but when to save the potential changes made by reader to flashmem?
and the powerdown...

bogiton, 2019-02-22:
That needs some thought. And definitely some testing.

closed 2019-08-17 by iceman1001