iceman1001/ChameleonMini-rebooted #24: Provide Precompiled Hex/Bin

WolfgangMau, 2018-01-20:
would it be possible to provide compiled Versions of the Chameleon-Mini.hex/Chameleon-Mini.eep to the releases?
Since at least for me, using OSX, the compiled files seems not to work (even if flashed with windooze) 
there are no errors while compiling, so it's kinda hard to figure out the problem (at least for me/ for now)

bogiton, 2018-01-20:
There shouldn't be any difference of the produced hex/eep files after re-compiling the code. So I'm not sure if there is a point to also provide those.
On the other hand, we could include the bin files (after running createbin in windows), but then again, you also have to be on windows to flash those with the BOOT_LOADER_EXE.

WolfgangMau, 2018-01-20:
it's not the problem that I have to use windows to get them on the chameleon, it's more about that the file I compile on osx did not work.
Therefor it would be nice to files that work for sure and which I can compare/diff my files against.

kgamecarter, 2018-03-30:
You can enable WSL and install ubuntu in Windows 10.
Compile work fine.

stek29, 2019-04-10:
They're now available on Releases page.

iceman1001, 2019-04-10:
time to close this one.

closed 2019-04-10 by iceman1001