iceman1001/ChameleonMini-rebooted #25: Functionality of MF_DETECTION

gtpy, 2018-01-20:
What is the functionality of the MF_DETECTION configuration?

What should be happen if a slot is configured in detection mode?

Thanks a lot.

iceman1001, 2018-01-20:
It saves the reader / card authentication sequence and it is used later to run mfkey32 with, in order to get the key used.

**The process**
Basically,  you set MF_DETECTION,    you present device to a genuine reader and wait for awhile.
Then you reconnect device to computer and press 

- **mfkey**  (in new GUI)  
- **fast calc**  (in old GUI)

and if you are lucky,  some keys are found and printed in textbox..

gtpy, 2018-01-20:
Thanks a lot.

You think it is possible to add this functionality to a RevG too?
I like my rebooted device, but RevG has, reader, sniffer, ... that I don't want to miss.
Anyone has ideas which files depending on mf_detection?

- /Application/Detection.c
- /Application/Detection.h
- /Application/Application.h
- /Terminal/Commands.c
- /Terminal/Commands.h
- configuration.c
- configuration.h

Special things to care about?

kevin2008-01, 2018-01-20:
But for working, you need one valid UID for the reader.

iceman1001, 2018-01-21:
Of course its possible,  everything you need you have. 
You still would need the GUI to do the cracking part,  or you need to implement the mfkey32 attack into the commandline part.

Just understand what you need to execute it.

iceman1001, 2018-01-21:
closing since its a question for RevG - repo.

closed 2018-01-21 by iceman1001

gtpy, 2018-01-21:
@iceman1001 thanks I understood.
I'll go on for it. 
Cheers