iceman1001/ChameleonMini-rebooted #29: Crypto patch

jackkleeman, 2018-02-07:
Hi guys, could you push the crypto patch to a branch? My device is totally useless without it

iceman1001, 2018-02-07:
there one big problem,  the crypto1 patch is too big for the 32kb we can play with.  The compiled firmware image doesn't fit and renders your device totally useless..

jackkleeman, 2018-02-07:
Can you put it as a branch anyway? I would prefer to remove lots of other features as I only need classic 4k.

Caydenz, 2018-02-17:
Is it possible to replace the ATxmega32A4U with a ATxmega128A4U and stay compatible with the original firmware (and this one as well)?

The only differences are the program memory, eeprom size and ram kbytes.

iceman1001, 2018-02-17:
good question.  Don't know if the GPIO pins are the same...  regrettfully I don't know what difference has been done to the pin mappings.

Caydenz, 2018-02-17:
It looks like all xmega A4U microcontrollers have the same pinout. ( http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8387-8-and16-bit-AVR-Microcontroller-XMEGA-A4U_Datasheet.pdf )

I'm going to give ATxmega128A4U a try when my lab401 order arrives and see how that works out.

iceman1001, 2018-02-17:
A missunderstanding,  the question is which GPIO pins has been connected on the PCB with the MCU.  Not if the GPIO pins are the same on the two MCU.

WolfgangMau, 2018-02-17:
at least it is not possible to run the firmware of the reve-rebooted on a revg ( without refactoring ) -I‘ve tested that

icsom, 2018-02-17:
additional note: of course I changed the mcu-type in the Makefile ;-)

iceman1001, 2018-02-28:
Got hold of @iskuri     He has optimized a crypto1 implementation before, so lets keep our fingers crossed!

bogiton, 2018-04-03:
Is there any progress on that? Should we try to add the existing (big) patch in the next release?

iceman1001, 2018-04-03:
nay,  the big patch takes all memory available...
you can always make a patched firmware version with it,  but I still have faith in @iskuri

bogiton, 2018-04-03:
hehe I put my faith in @Iskuri as well.
I should just be patient then for the new release! :P

Iskuri, 2018-04-03:
Don't worry guys I am still on it, unfortunately life has gotten in the way a bit but I have more free time to look at it this week and will be doing so.

iceman1001, 2018-04-03:
not the least worried,  I am just more eager to see it :)

Iskuri, 2018-04-06:
Hey guys, I've made some progress on this one, I've just got a few annoying bugs to iron out with it and then I'll do some cleanup and make a pull request. I've basically integrated a lot of my code that was optimised in AVR Asm into it and changed the core codebase to better work with this. This means that the Crypto1Byte and Crypto1Nibble calls have been updated to call an asm function called Crypto1Bit which does the filter and PRNG succession functionality that's usually performed by Crypto1FilterOutput and Crypto1LFSR. I've only changed things in Crypto1.c so far and created a new file for the asm functions called Crypto1_asm.S. I've not run any speed checks yet as I need to iron out the bugs but after I have I will work a bit more on any other optimisations. I'm also planning on hardcoding the PRNG'ed nonces in the parts where the nonce is already hardcoded as this will save a hell of a lot of time as well.

iceman1001, 2018-04-10:
I merged @Iskuri 's PR.   It compiles and runs.
@jackkleeman   you can test to see if your issues is solved with it?

jackkleeman, 2018-04-10:
Sure although my big problem is the broken code for the mifare classic 4k.

iceman1001, 2018-04-10:
@jackkleeman   one step at the time.  First test this one.

closed 2018-04-10 by iceman1001

jackkleeman, 2018-06-14:
This solved my speed of crypto problem