iceman1001/ChameleonMini-rebooted #31: No serial output other then 200 from terminal ?

prensel, 2018-02-21:
I'm trying to use terminal access using putty but all i get is the 200:UNKOWN COMMAND response.
I cant even check what firmware currently there is on it because uploading any firmware other then supplied by lab401 doesnt work.
All information i have found sofar is rather incorrect or for the revG version (different mcu)
Ok it does work as it is but when theres no decent way to compile, upload and config any new firmware its saddly of no use actually...

bogiton, 2018-02-21:
Maybe you are forgetting the "MY" extension of the commands for the rebooted version? It is HELPMY instead of HELP, VERSIONMY? instead of VERSION?, etc.
I believe that there is a good deal of information in our Wiki (https://github.com/iceman1001/ChameleonMini-rebooted/wiki). If some of it is incomplete as you say, please feel free to contribute.

iceman1001, 2018-02-21:
And for putty to work,  you would also need to enable  "Local echo" in order to see what you write

iceman1001, 2018-02-21:
The most of us here has successfully compile,  flashed and ran the chameleon revE with new GUI,  so your experiences differ quite a lot from us.

WolfgangMau, 2018-02-21:
I can just compile the firmware, but it’s not running on the device (
compiled on os x & flashed with windooze flash.bat)
I can also only get the precompiled version to run ... even if the
compilation and flashing  throws no errors

-- 

Wolfgang Mau

DevOps Engineer

MOIA - Social Movement
Axel-Springer-Plartz 3 / 20335 Hamburg / Germany
M +49 173 5302129
www.moia.io <https://moia.io>
<https://moia.io>

A Company of the Volkswagen Group

Post address: MOIA GmbH, c/o WeWork, Kemperplatz 1, 10785 Berlin
Managing Directors: Ole Harms, Dr. Frank Dilger, Robert Henrich
Phone: +49 30 726219941 / E-Mail: info@moia.io

Registration Court: District Court of Berlin-Charlottenburg
Registration Number: HRB 181035
Sales Tax (VAT) Registration Number: DE308624128

<https://www.linkedin.com/company-beta/22333228/>
<https://www.facebook.com/MOIAmobility/>
<https://twitter.com/MOIAmobility/>

iceman1001, 2018-02-21:
_most of us_

So you never got it to run?

prensel, 2018-02-21:
Ok, great the suffix MY did certainly help ! But where is that mentioned ?
I'm currently writing a small howto and will post asap.
Compiling on Debian 9 just works fine but biggest issue now is that neither avrdude or dfu-programmer cant connect properly wit the mcu.

iceman1001, 2018-02-21:
it was mentioned on the Wiki,  which I guess you didn't read before...

prensel, 2018-02-21:
The wiki for the RevG ? 
Yes i read that one even printed Chameleon Command Structure as a guide..

iceman1001, 2018-02-21:
No,  the wiki here where you posted, which is about the RevE rebooted version

bogiton, 2018-02-21:
https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Terminal-Commands
Here.

prensel, 2018-02-21:
Ok i see it now. Never noticed that one because the GettingStarted link is pointing away from this site to the revG one... Maybe it's better to change that ?

iceman1001, 2018-02-21:
true,  its taken from offical RevG,   I updated it to point to our wiki and be more clear.

closed 2018-02-21 by iceman1001