iceman1001/ChameleonMini-rebooted #32: Implementing CLONE function

prensel, 2018-02-21:
What would it take to implement the CLONE function as recently added to the RevG version ?
I've started looking at that piece of code to see how/what it takes but maybe someone else is doing that also ?

iceman1001, 2018-02-21:
Clone? Like how is it suppose to work? Are you thinking of using chameleon mini as a reader to program tags?  Not so sure that stuff will fit.  The revG has 128kb to play with,  revE rebooted has only 32kb.

bogiton, 2018-02-21:
I just noticed that new CLONE command they added in RevG. It uses the reader functionality of RevG (acts as reader to identify and then potentially clones), which haven't added here at all. And as iceman1001 notes, the available memory is very limited for all the features.

WolfgangMau, 2018-02-21:
but it's not that useful - it is just copying the uid - does it really worth it ...ok, you can copy the id and start a reader attack ... but you need to copy the uid to a slot which is configured for detection before anyway ... only/but needful for 'spontaneous attacks' 
or, add also a button-action which switches the configuration to mf_detection 

you (the coders) could also think about different firmware-version with different functions like 
- one version for raw communication, but no or less slots
- one for sniffing and detection ... maybe clone to one slot
- one for best of all worlds - like now
- one for ...
...
welcome to the maintenance hell :-)

... maybe it would be easier to implement the detection-feature to the RevG 👍

iceman1001, 2018-02-21:
Aint going down that road.   And the RevG ppl can implement what they want, its unrelevant to this repo.

iceman1001, 2018-02-21:
the copy of uid,  and set a slot for mfdetection, would be easy.  No need to write a tag.

iceman1001, 2018-03-20:
some update,  
I just heard that the RF circut on RevE rebooted is only made for simulation.  It can't not output power.

Clone function is not possible.

closed 2018-03-20 by iceman1001

bogiton, 2018-03-20:
Oh, that means that no reader support is possible on our device? No point in even the attempt of adding it?
I hope that at least it simulates the MF Classic tags properly :P

iceman1001, 2018-03-20:
Yup,  no point at all.