iceman1001/ChameleonMini-rebooted #34: Ultralight EV1 emulation problem

bogiton, 2018-03-17:
I was testing out a cloned UL EV1 tag on a real reader, but it seems that I face the same problem as the guy who implemented this in the first place on the RevG: https://github.com/emsec/ChameleonMini/issues/104#issuecomment-293476725
I have also enabled temporarily the logging functionality on our device, just to log the different authentication attempts in a single run, but, same as the other guy, nothing got logged, only the BOOT signal of the device (so, yeah, the logging works as it is).

Before the real test, I run through the full communication (which was previously sniffed with PM3) step-by-step with the PM3 and the responses of the Chameleon were a perfect copy of the original ones. Could this be a timing problem? or maybe some clone detection technique? Speaking of clone detection, I came across the following fix but I'm not sure if it could resolve the issue: https://github.com/emsec/ChameleonMini/pull/114/commits/d630363608770abfef641a22c0a4cd155a61d333
I'm open to any idea.

iceman1001, 2018-03-17:
that config / lock fix should make the device to behave better.
timings could be one major reason as well.

iceman1001, 2019-08-17:
time to close?

bogiton, 2019-08-19:
I can't test ULEV1 now. If you want, we can close it for now and re-open at some point if necessary.

closed 2019-08-19 by iceman1001