iceman1001/ChameleonMini-rebooted #36: Noob Walkthrough

robjbone, 2018-05-06:
Hi,

I'm pretty new to RFID testing, I've picked up a chameleon mini rebooted running **Chameleon-new-1.0** firmware from [lab401](https://lab401.com/products/chameleon-mini-reve-rebooted) and while it looks great, I'm struggling to get started on it.

I'm initially just looking to read Mifare Classic tags, but I'm only seeing commands for card emulation or reader attacks in [the commands](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Terminal-Commands) and "standard" chameleon guides obviously don't work for this firmware.

Could anyone help me with the fundamentals of this device? I'll be happy to write a noob walkthrough guide afterwards for the wiki and then [blog](https://jmpesp.me) about it.

Thanks,
m0rv4i

kevin2008-01, 2018-05-06:
please test this GUI by iceman.
http://www.icesql.se/download/ChameleonMiniGUI/publish.htm

iceman1001, 2018-05-06:
The terminal commands is just terrible.  I never even consider using them, which is strange since I do love playing with proxmark3 client.  Use a GUI.  There are quite a few out there now, I guess it supports the idea that the terminal commands is hard to use.   You are most welcome to blog about GUI, terminal commands etc or feel free to update the wiki here.

bogiton, 2018-05-06:
@robjbone Also bear in mind that the Chameleon-Mini rebooted is not suitable for reading tags, as iceman already mentioned in another issue. It can only be used for emulation.

robjbone, 2018-05-07:
Thanks for the swift responses!

To check my understanding then, I'd clone tags using, for example, the proxmark, and then upload the dumps to the chameleon? I've checked out the GUI and doing that seems clear.

Other than that it can do reader attacks, but not just read tags directly? Would this be MF_DETECTION mode? I'm unsure as to how i'd do this using the GUI.

Thanks again.

bogiton, 2018-05-07:
That is exactly right. Full dump of the tag using the proxmark and then emulate using the Chameleon. You would only need to convert between the proxmark dump format to the Chameleon one, before uploading the dump using the GUI.

About the reader attack, the steps are the following:
- Using the GUI set a slot to MF_DETECTION
- Approach the Chameleon to a MFC reader and let it work for a few seconds
- Back to the GUI, click on the mfkey32 button to start the cracking process

Hope that helps.

iceman1001, 2018-05-07:
...yeah that converting of dumpfiles would be nice to some functionality in GUI...

iceman1001, 2018-05-07:
yes, the reader must start a authentication in order for it to work..

and yes,  a user-friendly way of converting.. I already have some templates which shows a pm3 dump  or the offset for the special data in the beginning.

dunderhay, 2018-07-27:
Hi there,

I noticed that the `identify` command has not been implemented and seems to be commented out in lines 275-281 of the terminal commands : https://github.com/iceman1001/ChameleonMini-rebooted/blob/master/Firmware/Chameleon-Mini/Terminal/CommandLine.c

```
  {
    .Command	= COMMAND_IDENTIFY_CARD,
    .ExecFunc 	= CommandExecIdentifyCard,
    .ExecParamFunc = NO_FUNCTION,
    .SetFunc 	= NO_FUNCTION,
    .GetFunc 	= NO_FUNCTION
  },
```
Is there a way to read the atqa, uid, and sak with identify command like Rev,G ?

Cheers in Advance

iceman1001, 2018-07-27:
Apperantly there shouldn't be any reader functionality in RevE Rebooted,  only simulation,  hence all 14a-reader functionality in offical RevG is useless to us.

Pilly170, 2019-01-09:
I have another noob question, If you choose the Random_UID on the button, how do you move to another tab.

bogiton, 2019-01-10:
@Pilly170 You could set the Button Long function to SWITCHCARD to move to the next tag (by pressing the button for 1-2 sec).

iceman1001, 2019-01-10:
use the BUTTON dropdown instead...

bogiton, 2019-01-10:
@Pilly170 or you have to flash a newer firmware :)

Pilly170, 2019-01-10:
still ghosted, looks like i have firmware 1.0, are there any walkthroughs on how to flash? 
I also dont have the output view at the bottom, but then again Im running it on a VM and the res. maybe out.

bogiton, 2019-01-11:
@Pilly170 I think this guide in our Wiki is clear enough: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing---windows
On the Serial tab of the Chameleon GUI, is the BUTTON_LONGMY command there in the list when you connect your device?

olaf1234, 2019-01-16:
Hello,
For Mf detection does it work with only ud1,ud2,ud3,BCC1 (Mifare ultralight). Can't upload the full UID (seems to be made for MF classic).

bogiton, 2019-01-16:
@olaf1234 MF_DETECTION is only applicable for Mifare Classic tags.

olaf1234, 2019-01-17:
Ty for your anwser @bogiton 

One question lead to another:

Some context : When i'm reading my MF0ICU1 emulation  with my NFC taginfo app, the card UIDS,BBCS, OTP, LOCKBITs, DATA, manufacturer, RF technology, ATQA and SAK are identical to the real one. I uploaded on the chameleon a .bin file of my dump. Althought, even if the access conditions are read correctly (i got 3 read only pages), I can still modify them (include OTP and lock bits) with my MIFARE++ ultralight app. When i present my chamelon emulation to the real reader/writer : Incompatible card.

Questions: Does the state of my chameleon could be the problem(ex : IDLE, Halt)? , Does the real access conditions ( not the ones read) is noticed by the real reader/writer. Could it be something else? If those questions are valid interrogations(or not lol), can anyone give me some hints to resolve this? I'm pretty new to this.

iceman1001, 2019-01-17:
Keep to subject.  Don't hijack issues.  Open a new issue related to the software.

closed 2019-01-17 by iceman1001

yaxax, 2019-03-11:
Can somebody explain to me about the reader attack chameleon mini reve.1 you select a slot.2 mf_detect.3 swichcards.4aply.5 then put the chameleon mini reve on the reader for a few seconds.is that the right way? And the uid function must be blank ?or ffffffff ?are with a valid uid?

bogiton, 2019-03-11:
Hey @yaxax 
The MF_DETECTION configuration actually emulates a MiFare Classic 1k card against a valid reader.
Your steps seem correct. Set a slot to MF_DETECTION configuration, place it on the reader for a few seconds and then press the mfkey32 button on the Chameleon RevE rebooted GUI to retrieve the key.
As for the UID, it depends on the reader I guess, but it is more probable to work with a valid UID.

iceman1001, 2019-03-11:
...but he would need to run the latest firmware from this repo,  or the default one...   and the latest GUI aswell...

yaxax, 2019-03-11:
I have bought the chameleon mini reve at lab401 and have downloaded the gui 1.1.0.1 trough  their web and I also have the 1.2.0.18 iceman edition but can not read the output the  gui is bigger than my screen.And how can i put the info on a card like the video using a proxmark3?

yaxax, 2019-03-11:
I allso have the chameleon mini app is that allso the latest version gui?And 1.1.0.1 is that allso the latest version gui

iceman1001, 2019-03-11:
no... v1.2.0.18  is the latest

yaxax, 2019-03-11:
I have the 1.2 .0.18 version but i can't see are read output.the gui is bigger then my screem I don't know how to explained this because I'm dutch and english is not so good but do you understand what I mean that I can read the output can you send me a link where the gui comes normaal on mij screem

yaxax, 2019-03-11:
When I do mfkey do i get 32 numbers are 12 numbers ? im new in this just learning.

iceman1001, 2019-03-11:
32numbers   ,  can you post the output?

yaxax, 2019-03-12:
I try to send a pic but I don't know how?

yaxax, 2019-03-12:
And the 32 numbers is that the block number 0?Are is it random block numbers?Because a block got 32 numbers right?

yaxax, 2019-03-12:
[image]
`

yaxax, 2019-03-12:
This I mean that I can't see the output off the gui on  my screem

iceman1001, 2019-03-12:
Let me see if we can make the main window resizable..  @bogiton

yaxax, 2019-03-12:
Ok thanks

bogiton, 2019-03-12:
@iceman1001 yeah, I am on it. Gonna try a quick fix with scrollbars.

iceman1001, 2019-03-12:
So Bogiton just added scrollbars,

iceman1001, 2019-03-12:
And I just push out  v1.2.0.19,    So restart your gui and you should get the latest....

yaxax, 2019-03-12:
[image]
Is still the same I can't read the output and the scrollbars don't work see foto

iceman1001, 2019-03-12:
You need to resize the window...

yaxax, 2019-03-13:
The gui still not working good see no output and I resize the window many time but still the same.But the older version works fine allso my other gui works fine.The latest gui most be fix.

iceman1001, 2019-03-13:
your description of the problems lack details.  On my computer the GUI is resizable and when I do resize it I get scrollbars which lets me scroll down to the output window.    I can't verify your problems.

yaxax, 2019-03-13:
I don't understand the output numbers i think keys?but i only see 12 numbers?instead of 32?
1 I put my uid in slot 1.2swichtcard.3aply.4 I disconnect the chameleon mini reve from the laptop and put it on the reader.
[image]
This are the results when I put the chameleon mini reve on the reader .But I don't know how to go further.
[image]And this is my card .

gtpy, 2019-03-13:
> [20190312_201715](https://user-images.githubusercontent.com/48149681/54267601-d0537300-4579-11e9-8037-28c10d91ac9a.jpg)
> [20190313_101821](https://user-images.githubusercontent.com/48149681/54267828-350ecd80-457a-11e9-95b5-e838da4792fb.jpg)And this is my card .

For me the GUI is working. And output is visible.

-------- Original-Nachricht --------

iceman1001, 2019-03-13:
When I look at the pictures,  the GUI is working well.  The output is visible.   

Given that the mfkey finds keys, I say that the firmware is working well.   So I suggest you google/youtube/read on pm3 wiki,  forum  to figure out what you need to do next.

**hint** yes those are keys...

yaxax, 2019-03-13:
I search but I can't find nothing what I have to do futher.I allso read prox forum and github forum.but I can't find what I have to do next .i don't understand the output and how to go futher whit the output.

yaxax, 2019-03-13:
And is the old version gui that's works fine but not the new version

iceman1001, 2019-03-13:
This is not the place to discus that.  On Github issues we discus bugs/features in the code.

yaxax, 2019-03-13:
But I talk about the chameleon reve I try to understand and learn my bad if I ask question that are related to the chameleon reve

yaxax, 2019-03-13:
The chameleon-mini-reve-rebooted  where can I find some information  to understand the output info .I only see information about chameleon mini are g but not off reve.

bogiton, 2019-03-13:
@yaxax The output is really straight-forward, if you know what you want to do. If you can't understand what "Key A" or "Key B" means, then you have to read the [datasheet](https://content.gototags.com/wp-content/uploads/Mifare_Classic_1K_MF1S503x.pdf) first. Also, since you have also a pm3, as far as I can see, you should be able to do what you probably need with that too... So, it's not a matter of Chameleon as a device at all.

yaxax, 2019-03-13:
And I see 2 different access bytes I mean

yaxax, 2019-03-13:
This is a better picture to understand what  I mean
[image]

yaxax, 2019-03-13:
When I put my uid in the gui and then put the chameleon reve to the reader i got my own card information so do I must put my uid in the gui?1 slot 2 uid?3 swichcards. 4 aply 5 put to the reader right.?Because their is not so information I see video but this guy don't explain it good don't now wat he meaning and on the forum their allso nothing so can somebody help me out all what ask s related to the chameleon

yaxax, 2019-03-13:
I try to learn but see no good info

yaxax, 2019-03-18:
When i  put the uid number at section uid and then press apply then the uid goes back to ffffffff why is that something wrong with the software?

bogiton, 2019-03-18:
Which UID are you trying to set? If the configuration is MF_CLASSIC then you have to enter a 4-byte UID (that is 8 hex digits). Usually, when the UID is of incorrect length it goes back to FFFFFFFF.

yaxax, 2019-03-20:
Hi I try the uid off a ultralight card and then i put the 8 numbers off the ultralight card but then its set up to ffffffff again in the uid section when i do mf_detect.and whit 8 hex digits you mean the 8 numbers off the uid?

bogiton, 2019-03-20:
@yaxax First set the configuration to MF_ULTRALIGHT and press apply. The UID should switch to a 7-byte one (14 hex digits). Then set the 7byte UID you want and press again the apply button.

bogiton, 2019-03-20:
Also, MF_DETECT is for Mifare Classic 1K only. Not for Ultralight.

yaxax, 2019-03-21:
Is their allso a way to detect ultralight?

[9 more replies]