iceman1001/ChameleonMini-rebooted #37: RevE Boards Now Supported by ChameleonMiniLiveDebugger App

maxieds, 2018-05-12:
I finally had a user inquire about RevE board support in the [Chameleon Mini Live Debugger](https://github.com/maxieds/ChameleonMiniLiveDebugger) Android app which provides a portable user-friendly GUI for the Chameleon boards. In addition to adding the USB vendor and product IDs which are specific to the RevE firmware on this site, I have added support for the button functionality for the RevE boards and [specific command variants](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Terminal-Commands) not found with the RevG boards. The [most recent](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger&hl=en_US) Play Store version (0.4.1-free) of the app has these updates built in. Recent screenshots (I'll include two below) can be found [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/issues/9). 

[image][image]

iceman1001, 2018-05-12:
Nice work,   

As mentioned in our email exchange,  RevE rebooted firmware still doesn't support the logging which is used on RevG offical firmware.  So I don't know how much usefulness it will be for people who runs the firmware from this repo in regards to _live debugger_ functionality but as a Android GUI for RevE it will be a nice addition.

closed 2018-05-12 by iceman1001

maxieds, 2018-05-12:
@iceman1001 :Would you mind putting this thread back on the main page so more of your users will see it? Perhaps as an enhancement?

iceman1001, 2018-05-12:
Its not an issue for this firmware.   If you want to add something to the wiki , feel free to do so.

maxieds, 2018-05-12:
Is this to say that the hardware cannot handle the logging capabilities, or
just that it hasn't been implemented in firmware yet? If it's the latter
I'd like to work on this as a project once I get my revE board. I'm already
working on another revG firmware mod to add rudimentary Desfire support, so
this would be a good warm up coding session for me.

iceman1001, 2018-05-12:
the hardware can handle it but you will need to rewrite / design / make shortcuts in number of possible enumerable cards because of the smaller flashmem available.  But definitly possible and a great warm up coding session for you

iceman1001, 2018-05-12:
Still,   if you want to add a line or two in the readme.md  about your android gui in order to promote it,  go ahead.

maxieds, 2018-05-12:
Will you include the patches in the main branch on your GitHub page if I
can get the logging functionality working for the revE boards? I have a
brand new revE coming in the mail soon. I love coding around with stuff
like this, and I think I can figure out how to make it work.

kevin2008-01, 2018-05-12:
sir..... https://www.youtube.com/watch?v=M_3O7lL4Us8
Keep cool.....

maxieds, 2018-05-12:
Probably a lot of cut/paste from the revG firmware source, but still.

maxieds, 2018-05-12:
@kevin2008-01 : Not sure what you mean by the kung fu panda video, but unless there's a hardware restriction on how fast the chip can read over RFID or something, this pretty much just seems like cut and paste from the RevG firmware source (+/- some buffering from the chip that hasn't been coded up in the RevE firmware yet). Semi-advanced microcontroller programming in C, yes? If as @iceman1001 suggests that it's possible, then I just need to get my hands dirty in some firmware code.

maxieds, 2018-05-12:
> Since I suspect serious mem-changes, I would consider making a branch with
> logging possibilities and one master branch as-is.
> There are some logging code from RevG inside already, but its not is use
> at all. But yes, the logging functionality would be some cut/paste from
> RevG and then afterwards the adaptations regarding memory space.

Cool. I will get started in the project once I get my RevE board in the
mail later this/next week.

BTW, has anyone successfully tested out the current version of my app (on
play store in both flavors) with the RevE boards? I suspect it now works
with recent firmware, but my user was unable to confirm. Either way, I will
get support for the board ironed out once it arrives stateside this week.

kevin2008-01, 2018-05-12:
what is is you app's, link   see you.

maxieds, 2018-05-12:
You can get the "unofficial" (but still signed APK) releases from here:
https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/0.4.4-free.
And the official Play Store releases from your phone's "official" software
here:
https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger&hl=en_US.
(It's the same APK re-signed by Google.)

kevin2008-01, 2018-05-12:
you need translation in french ?

maxieds, 2018-05-12:
Possibly? Do you have one?

kevin2008-01, 2018-05-12:
i'm ready. just do it.....

maxieds, 2018-05-12:
Where would I post it on Play Store? Do you know the interface for this?
Strangely enough my one RevE user was from France...

iceman1001, 2018-05-12:
Do please take this discussion over @maxieds  's repo instead of here.

kevin2008-01, 2018-05-12:
ok we go. see you, and thank for the future iceman1001
...

maxieds, 2018-05-12:
Just to be precise, I do not want to go into all of the details of the RevG
board's LOGMODE=MEMORY command. I just want to have the functionality of
LOGMODE=LIVE (see:
https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_log_8h.html#a34112fbd78128ae58dc7801690dfa6e0),
which more or less just prints the RFID-tagged logging data to the USB
serial terminal as it receives it. That is to say, I do not want to bother
with figuring out where to store backlogged logs into the small chip
onboard the RevE boards. I only want to replicate the LIVE logging feature
onboard the RevG devices. Of course this behavior should be able to be
turned off with a setting.