iceman1001/ChameleonMini-rebooted #38: anti copy, all new reader, fuck my chameleon !!!!

kevin2008-01, 2018-05-14:
Now, new reader, fuck the chameleon, with anti-copy tag system.....
Please implement enable / disable fuid uid...for slot..... if is possible, right. ???
Thank you iceman1001.

ceres-c, 2018-05-15:
I am not sure I understand what I'm reading...

iceman1001, 2018-05-15:
Enable / Disable FUID ?   UID for slot..

I'm not sure either but that could be of the french way of expressing things.  There seem to be a wish for FUID to be optional per slot.   What I'm curious is what functionality FUID stands for?  use once?  magic? ...

iceman1001, 2018-05-15:
Hm,  this firmware as it is shouldn't allow for uid (block0) re-writing.  
it should set uid when we upload a dump etc.  Then it shouldn't allow at all for block0.

Aha, I see in the source code there is no limit for block0 writes.   ok.  Fair enough, I agree.

ceres-c, 2018-05-15:
Now I get it and I agree

kevin2008-01, 2018-05-15:
the reader must understand that writing block 0 is impossible, he must just read the chameleon. I do not know which command the reader sends to do the test or detect a tag copy. apparently, new readers are not blocked only in France, but also in other countries. see you

quantum-x, 2018-11-30:
French readers implemented a 'magic check' where they send 0x43 0x40 to weed out magic tags.
However, the Chameleon Mini also has timing issues - which means they are often rejected by french readers. I'm assuming he's bumping up against problem 2 more than problem 1 - as the Chameleon should not respond to 0x43 0x40

slurdge, 2018-11-30:
By "magic", you mean Chinese backdoor command ?

quantum-x, 2018-11-30:
Correct.
Also - it seems that this behaviour has now been removed by default.

shinhub, 2019-08-14:
@iceman1001 to be closed I think.

iceman1001, 2019-08-14:
@shinhub  double check in code that block0 doesn't allow for Write commands...

closed 2019-08-14 by iceman1001

shinhub, 2019-08-15:
@iceman1001 you were right... I took quantum-x answer as clue of a previous patch.
I checked and found nothing in Mifare code to prevent Block0 write.
I am patching this.

iceman1001, 2019-08-15:
keep in mind for you patch,  that the support-chinese-magic command defines,  at compilation time,  should allow for updating block0 etc..

shinhub, 2019-08-15:
I swear won't hurt any chinese.

iceman1001, 2019-08-15:
You do have a sense of humor.  Excellent.

commit 28678d30 (2019-08-17): Merge pull request #120 from shinhub/atomic_fix_writable_block0

Fix #38: block0 was always writable
commit dfe56852 (2019-08-17): Fix #38: block0 was always writable