iceman1001/ChameleonMini-rebooted #39: Getting the RevE Board Back Into Bootloader Mode

maxieds, 2018-05-23:
I ran the first erase command (which works with the RevG boards) [here](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing-Linux-(Unix)): 
```
$ sudo dfu-programmer atxmega32a4u erase
```
This successfully erased the chip, but to the point that there's no longer any trace of the Chameleon program on it. I have read the discussion at #11 (not sure I completely understand all of its implications). I tried to reflash the board with the stock RevE .eep/.hex files using the Win32 tools in the repo, but the problem is that all I have is a 64-bit Windows so these EXE files do not run -- even in compatibility mode. I also modified the Batch script on the RevG site which uses dfu-programmer on Windows to flash this. No dice, no go with the flashing. 

When I try to run the dfu-programmer on Linux it says that the XMEGA chip is write protected. Does this mean I can only get my Chameleon back to sane state by using the Win32 flashing utilities in the repo on Windows? In otherwords, is the chip effectively locked without the use of the CreateBin.exe application. How do I proceed with getting the Chameleon back? I hope I didn't permanently brick the damn thing in the first two days I've had it to play with.
I ran the first erase command (which works with the RevG boards) [here](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Flashing-Linux-(Unix)): 
```
$ sudo dfu-programmer atxmega32a4u erase
```
This successfully erased the chip, but to the point that there's no longer any trace of the Chameleon program on it. I have read the discussion at #11 (not sure I completely understand all of its implications). I tried to reflash the board with the stock RevE .eep/.hex files using the Win32 tools in the repo, but the problem is that all I have is a 64-bit Windows so these EXE files do not run -- even in compatibility mode. I also modified the Batch script on the RevG site which uses dfu-programmer on Windows to flash this. No dice, no go with the flashing. 

When I try to run the dfu-programmer on Linux it says that the XMEGA chip is write protected. Does this mean I can only get my Chameleon back to sane state by using the Win32 flashing utilities in the repo on Windows? In otherwords, is the chip effectively locked without the use of the CreateBin.exe application. How do I proceed with getting the Chameleon back? I hope I didn't brick the damn thing in the first two days I've had it to play with.

maxieds, 2018-05-23:
As it happens, the reason I was re-flashing the device to begin with is to test out some logging firmware mods related to #37 from last week. The firmware files are located [here](https://github.com/maxieds/ChameleonMini-rebooted/releases/tag/0.2.0-logging) for testing. Pretty basic add-ons to the source, I just can't test them out at the moment!

iceman1001, 2018-05-23:
yeah, the linux flashing isn't solved, meaning we haven't figured out how to do it. :(
There were some discussions on issues #11  #12   It tooks some time before we managed to flashed the compiled firmware in this repo using the tools needed.   We wrote it down in the wiki aswell.

But basically,  figuring out a way to solve the linux flashing would be great.

maxieds, 2018-05-23:
So is the issue that the manufacturer locked the chips except for with these bootloader flashers? Will these utilities restore the Chameleon firmware on my device?

iceman1001, 2018-05-23:
Dunno actually about locked chip, we only know bits and pieces in the flashing process.
Classic lockdown the firmware in order to make money, sadly,  having a closed firmware also means high demands for bugfixes etc from the manufacturer which didn't happen.  Almost kill off the rebooted product.

If you use the original firmware files with the windows tools,  you will get a working rev E rebooted again.

shinhub, 2019-08-14:
@maxieds please try reading Wiki as it has been recently reworked. The Makefile also includes a convenient `make dfu-prog` if needed. @iceman1001 due to age I guess this could be closed.

closed 2019-08-14 by iceman1001