iceman1001/ChameleonMini-rebooted #41: Memory collision issue

Skydev0h, 2018-05-24:
It seems that after commit https://github.com/iceman1001/ChameleonMini-rebooted/commit/459d9aa90e28eadab68d001c6b9aa4bed76bb1d8 there is memory collision issue.

First, the code makes no sense: 
```	
if ( GlobalSettings.ActiveSetting == 0 )
		FlashAddress = Address + (uint16_t) GlobalSettings.ActiveSetting * MEMORY_SIZE_PER_SETTING_4K;
	else
		FlashAddress = Address + (uint16_t) GlobalSettings.ActiveSetting * MEMORY_SIZE_PER_SETTING_1K;
```

For the first case GlobalSettings.ActiveSetting is always zero.

Second, if the first slot uses 4K bytes (0..4095), then with this code second slot will clash into 1024..2047, third into 2048..3071 and fourth into 3072..4095, therefore writing to 4K slot 0 will damage/erase slots 1-3, and vice versa.

Logically correct way of determining address if you want first slot to be 4K and others to be 1K is:
```
if ( GlobalSettings.ActiveSetting == 0 )
		FlashAddress = Address;
	else
		FlashAddress = Address + MEMORY_SIZE_PER_SETTING_4K + (GlobalSettings.ActiveSetting - 1) * MEMORY_SIZE_PER_SETTING_1K;
```

(By the way you can drop the casts because Address is now already uint16_t).

What do you think about it?

By the way, what is the size of the SPI memory that we have?
I can't determine it by looking on chip because something is done with top of it (partially removed?)

iceman1001, 2018-05-24:
several questions in one..
yes,  the 4k first slot limit overlaps the rest.  A change is needed indeed.
the flashmemory should be 32kb, so I don't understand your edited comment.  I see no reason for a manufacturer to lie about its actually bigger than it is..  Since that would have been a advantage.

Skydev0h, 2018-05-24:
I could not find information about external flash size in the Internet, so i attempted to get the information from the chip directly. It may report density differently than standard code (however, 1F is manufacturer byte for Atmel, so that it may raise some questions). I will try to do out-of-bounds writes and reads and try to check for real capacity of chip. It may indeed be a "china version" of atmel chip that reports A but actually is B.

Skydev0h, 2018-05-24:
I have attempted to address high memory addresses close too 512K limit by changing the following stuff:
```SPITransferByte( 7 ); // 7 * 65536``` in FlashRead
```uint16_t PageAddress = Address / MEMORY_PAGE_SIZE + 1792; // 7 * 65536 / 256``` in FlashWrite

iceman1001, 2018-05-26:
@Skydev0h  are you doing a PR for the memory overlap?

shinhub, 2019-08-11:
This seems to be still an issue (that I unfortunately just discovered #22).
At this point I do not understand how is 4k supposed to be handled in memory, and how is our "SPI" memory different than for RevG.

commit 55030a58 (2019-08-12): Merge pull request #110 from shinhub/4k_slot1_memory_overlap_fix

4k memory overlap fix #41 + SPI_FLASHINFO rewrite
commit 7d60b6d6 (2019-08-12): 4k memory overlap fix #41 + SPI_FLASHINFO rewrite
closed 2019-08-12 by iceman1001