iceman1001/ChameleonMini-rebooted #44: problem sniff MF-Detect, OK with Official App, but not with Iceman Fork

Shashadow, 2018-06-03:
Hello,

Just a little message about a little problem I have with sniff (MF_DETECT)
I noticed for my last sniff I didn't have any output with iceman app (mfkey32), while with very old official release app I have the key (full calc)

[image]
[image]

maybe it's normal, I don't know ? do we have an option for a full mfkey32 maybe ? I didn't find.
Thanks
Shashadow

Shashadow, 2018-06-03:
oppps, sorry I was wrong to post here, I confused with :
https://github.com/iceman1001/ChameleonMini-rebootedGUI
my bad :-(

iceman1001, 2018-06-03:
yeah, different repo.  Try the latest GUI code to see if it works better.
In iceman gui,  there is no **fast calc** **full calc** only **mfkey** which looks on all existing collected data like full calc did.
Did you just try to swap between GUI applications or did you change anything with the application/chameleon..

Shashadow, 2018-06-03:
it's the same with lastest gui code, no more key, and I verified with official soft, it's Always ok after "full calc"

[image]

I just swap between app, I didn't make no change on chameleon, here output for versionmy
--> VERSIONMY?
101:OK WITH TEXT
Chameleon-new-1.0

iceman1001, 2018-06-03:
so using the old gui,  it solves keys?
using the latest iceman gui,  it doesn't solve anything..
Hard one to debug.  We did update the mfkey implementation.  Maybe a debug output to get which values was used for mfkey could be useful.  It should also take some time to run it. Since it will run multiple mfkey sessions in order to find keys.

iceman1001, 2018-06-03:
Sector 13, Block 52.  It shoulds two different keys for same  block which is a indication for failure.  Which is possible for mfkey.    Can you cut and paste all text from that "key detection result" box?

Shashadow, 2018-06-03:
I just have 2 discovery key with official app, not more.
the good key is the second for my tag (Noralsy tag, if you want add to defaut_key file to PM3 no problem (because no include in default_keys) and second key (after hardnested) is 4243414F5250) :-)
13sec52blo A:
314B49474956
13sec52blo A:
4143414F5250

Maybe reader has 2 differents keys for tag and try both together ? 
From my memory, I already use chameleon with your app, but with a another reader card, and it worked, but I had only one key, so maybe it's a good indication for failure...

Shashadow, 2018-06-05:
just for add a comment, I sniff another reader on slot 2, the results :

-- with iceman fork --

[Tag slot 1]
nothing...

[Tag slot 2]
[S0] KeyA [a0a1a2a3a4a5] 
[S0] KeyA [a0a1a2a3a4a5] 
[S60] KeyA [484558414354] 
[S60] KeyA [484558414354] 

-- with official app --

slot 1 :
13sec52blo A:
314B49474956
13sec52blo A:
4143414F5250

slot 2 :
15sec60blo A:
484558414354
0sec0blo A:
A0A1A2A3A4A5

It's not really a matter for me, just it's strange it doesn't success to calc key in the slot 1
if ok we can close this issue, if iceman fork doesn't give me a output just for 1 or 2 times, i just check with original soft, and no problem.

iceman1001, 2018-06-06:
No that didn't make any sense. Slot 1 doesn't work but slot 2 does...
Would you mind testing the other slots aswell?  To rule out that there is more slots not working.

I do spot another wrong,  S60 should have been S15..

iceman1001, 2018-06-07:
First a minor fix for GUI.
https://github.com/iceman1001/ChameleonMini-rebootedGUI/commit/8ea3f5c90061edeeb49c45a17623aaa2d34bddc1

The output showed sector instead of block. 

Then a fix for firmware..
https://github.com/iceman1001/ChameleonMini-rebooted/commit/cfd6a5d569e115ec74bfc8219b45ef41bc2fb7f1

the calculation when zero, was wrong.  However it still is a problem with were the collected data is saved. 

@Shashadow  would you mind testing things out?

Shashadow, 2018-06-07:
just for add : 
when I try to read with my phone the chameleon with MF_DETECT, it was the same than reader, no reaction.

I think either I did a bad thing when a upgrade firmware (but I  Don't think because emulation works, and I can see the 3 ultralight option (EV), either your firmware is faulty.

So I used "upgrade" and "load défauts" with your app, this seems restore original firmware, and Indeed already official software works again. and when I use my phone with MF_DETECT, the phone react, so I think there a something wrong in new firmware.

iceman1001, 2018-06-19:
Sorry to say,  but there is no time left for me to work on it.  Many interesting things could be adjusted and fixed for this Chameleon mini firmware, making it really usable in the field.

iceman1001, 2019-03-07:
did you fix this one @bogiton ?

bogiton, 2019-03-07:
I hope so! I need someone to confirm. I just tried it with my pm3 with a test auth key and then I managed to recover successfully that key from within our GUI.

iceman1001, 2019-03-08:
Ok it seem like the latest firmware and latest GUI sorts this out. @Shashadow  you can verify if you want but others already has done it.

closed 2019-03-08 by iceman1001