iceman1001/ChameleonMini-rebooted #46: Chameleon mini Chinese version (Linux)

TITANForze3D, 2018-06-10:
greet all friends. my english is bad. why can not I update the firmware. 

`$ lsusb`

> Bus 006 Device 004: ID 03eb:2044 Atmel Corp. LUFA CDC Demo Application

`$ sudo bash`
`$ socat - /dev/ttyACM0,crnl`
`UPGRADEMY`
`$ lsusb`

> Bus 006 Device 006: ID 03eb:2fe4 Atmel Corp. ATxmega32A4U DFU bootloader

`/dfu-programmer$ sudo dfu-programmer atxmega32a4u erase`

> Checking memory from 0x0 to 0x7FFF...  Not blank at 0x7FF1.
> Erasing flash...  Success
> Checking memory from 0x0 to 0x7FFF...  Not blank at 0x7FF1.

`/dfu-programmer$ sudo dfu-programmer atxmega32a4u flash --eeprom Chameleon-Mini.eep`

> Checking memory from 0x0 to 0x3F...  Empty.
> 0%                            100%  Programming 0x40 bytes...
> [ X  ERROR
> Memory write error, use debug for more info.

`dfu-programmer$ sudo dfu-programmer atxmega32a4u flash Chameleon-Mini.hex`

> Checking memory from 0x0 to 0x5DFF...  Empty.
> 0%                            100%  Programming 0x5E00 bytes...
> [ X  ERROR
> Memory write error, use debug for more info.

what can these error data mean?
how can they be eliminated?
help me please.

ceres-c, 2018-06-10:
Have you tried to...

> Memory write error, **use debug for more info**.

?

Add this option to both flash commands `--debug 100`

TITANForze3D, 2018-06-10:
no, have not tried

 `/dfu-programmer$ sudo dfu-programmer atxmega32a4u flash Chameleon-Mini.hex --debug 100`
      target: atxmega32a4u
      chip_id: 0x2fe4
      vendor_id: 0x03eb
      command: flash
      quiet: false
      debug: 100
      device_type: XMEGA
------ command specific below ------
   validate: true
   hex file: Chameleon-Mini.hex

atmel.c:1153: atmel_flash( 0x7ffdb2ae2290, 0x7ffdb2ae2180, false, false )
intel_hex.c:673: intel_flash_prep_buffer( 0x7ffdb2ae2180 )
atmel.c:1193: Flash available from 0x0 to 0x7FFF (64kB p. 0 to 0), 0x8000 bytes.
atmel.c:1197: Data start @ 0x0: 64kB p 0; 256B p 0x0 + 0x0 offset.
atmel.c:1201: Data end @ 0x5DFF: 64kB p 0; 256B p 0x5D + 0xFF offset.
atmel.c:1206: Totals: 0x5E00 bytes, 94 256B pages, 1 64kB byte pages.
atmel.c:819: atmel_blank_check( 0x7ffdb2ae2290, 0x00000000, 0x00005DFF )
atmel.c:927: atmel_select_memory_unit( 0x7ffdb2ae2290, 0 )
atmel.c:959: Selecting flash memory unit.
Checking memory from 0x0 to 0x5DFF...  
atmel.c:986: atmel_select_page( 0x7ffdb2ae2290, 0 )
atmel.c:1000: Selecting page 0, address 0x0.
atmel.c:752: __atmel_blank_page_check( 0x7ffdb2ae2290, 0x00000000, 0x00005dff )
atmel.c:784: Flash region from 0x0 to 0x5DFF is blank.
atmel.c:862: Flash blank from 0x0 to 0x5DFF.
Empty.
atmel.c:927: atmel_select_memory_unit( 0x7ffdb2ae2290, 0 )
atmel.c:959: Selecting flash memory unit.
Programming 0x5E00 bytes...
atmel.c:986: atmel_select_page( 0x7ffdb2ae2290, 0 )
atmel.c:1000: Selecting page 0, address 0x0.
atmel.c:1289: Program data block: 0x0 to 0x3FF (p. 0), 0x400 bytes.
atmel.c:1424: __atmel_flash_block( 0x7ffdb2ae2290, 0x7ffdb2ae2180, false )
atmel.c:1386: atmel_flash_populate_header( 0x7ffdb2ae1c60, 0x0, 0x3FF, false )
atmel.c:1340: atmel_flash_populate_footer( 0x7ffdb2ae1c60, 0x7ffdb2ae20a0, 65535, 65535, 65535 )
atmel.c:1485: atmel_flash: flash data dfu_download failed.
atmel.c:1487: Expected message length of 1104, got -9.
atmel.c:1292: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:514: Error writing memory data. (err -4)

TITANForze3D, 2018-06-10:
`dfu-programmer$ sudo dfu-programmer atxmega32a4u flash --eeprom Chameleon-Mini.eep --debug 100`
      target: atxmega32a4u
     chip_id: 0x2fe4
      vendor_id: 0x03eb
     command: flash
      quiet: false
      debug: 100
device_type: XMEGA
------ command specific below ------
   validate: true
   hex file: Chameleon-Mini.eep

atmel.c:1153: atmel_flash( 0x7ffd35209530, 0x7ffd35209420, true, false )
intel_hex.c:673: intel_flash_prep_buffer( 0x7ffd35209420 )
atmel.c:1193: Flash available from 0x0 to 0x3FF (64kB p. 0 to 0), 0x400 bytes.
atmel.c:1197: Data start @ 0x0: 64kB p 0; 32B p 0x0 + 0x0 offset.
atmel.c:1201: Data end @ 0x3F: 64kB p 0; 32B p 0x1 + 0x1F offset.
atmel.c:1206: Totals: 0x40 bytes, 2 32B pages, 1 64kB byte pages.
atmel.c:819: atmel_blank_check( 0x7ffd35209530, 0x00000000, 0x0000003F )
atmel.c:927: atmel_select_memory_unit( 0x7ffd35209530, 0 )
atmel.c:959: Selecting flash memory unit.
Checking memory from 0x0 to 0x3F...  
atmel.c:986: atmel_select_page( 0x7ffd35209530, 0 )
atmel.c:1000: Selecting page 0, address 0x0.
atmel.c:752: __atmel_blank_page_check( 0x7ffd35209530, 0x00000000, 0x0000003f )
atmel.c:784: Flash region from 0x0 to 0x3F is blank.
atmel.c:862: Flash blank from 0x0 to 0x3F.
Empty.
atmel.c:927: atmel_select_memory_unit( 0x7ffd35209530, 1 )
atmel.c:959: Selecting eeprom memory unit.
Programming 0x40 bytes...
atmel.c:986: atmel_select_page( 0x7ffd35209530, 0 )
atmel.c:1000: Selecting page 0, address 0x0.
atmel.c:1289: Program data block: 0x0 to 0x3F (p. 0), 0x40 bytes.
atmel.c:1424: __atmel_flash_block( 0x7ffd35209530, 0x7ffd35209420, true )
atmel.c:1386: atmel_flash_populate_header( 0x7ffd35208f00, 0x0, 0x3F, true )
atmel.c:1340: atmel_flash_populate_footer( 0x7ffd35208f00, 0x7ffd35208f80, 65535, 65535, 65535 )
atmel.c:1485: atmel_flash: flash data dfu_download failed.
atmel.c:1487: Expected message length of 144, got -9.
atmel.c:1292: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:514: Error writing memory data. (err -4)

ceres-c, 2018-06-10:
Given I don't own this device I can't help you further, but I suggest you to try and program it under windows, as this board was reported to work better with MS'os than linux.

TITANForze3D, 2018-06-10:
OK. the sniffing mode is included? MF_DETECTION
for example to read the authorization of the card Mifare Plus (DESFire) will be able to?

ceres-c, 2018-06-10:
Recently a problem about sniffing was reported (https://github.com/iceman1001/ChameleonMini-rebooted/issues/44), but it'll probably be fixed sooner or later. I am not personally working on it, nor I know the priority this bug has in @iceman1001's todo list.

Mifare plus should be supported in backward compatibility mode, but DESFire is a whole different beast. To my knowledge there are no pratical attacks against it, being it online or offline. Correct me if I'm wrong.

TITANForze3D, 2018-06-10:
thank you friend. also tried to connect to the USB OTG and manage the ChameleonMini Live, but for some reason it did not work. something with the usb configuration. just started to think about this topic. will have to install windows)

ceres-c, 2018-06-10:
In the past I had problems with a proxmark clone and USB OTG on android, the USB interface was not enumerated by the kernel at all. You could try and install a terminal emulator like termux and see what dmesg outputs when you connect your chameleon. Honestly, since you have problems with android too, your chameleon could be bricked or something... I don't know honestly.

Report here how your testing in windows went, once you're done, please.

TITANForze3D, 2018-06-10:
OK. and you can tell the software / programs for reading / editing Mifare Plus under. mfoc and mfcuk only for Mifare Classic?

ceres-c, 2018-06-10:
mfoc and mfcuk work with Mifare Classic only since those exploit vulnerabilities later fixed with Mifare Plus.

To attack Plus you have to sniff the first key (you can do it with the chameleon or with the proxmark) and then use the hardnested attack to extract other keys. I don't know if there are working implementations of this attack made through an ACR122U or the PN532. You have to search on github, or buy a proxmark :-)

TITANForze3D, 2018-06-12:
Hello,
OS: Windows Ultimate x86 (and Kali Linux x64)
Install Programms: Cygwin, Atmel Studio 7
How to do the next step: Add the bin path of Cygwin and AVR Toolchain to the PATH environment variable.
Restore the firmware from the disk turned out. Have not tried to update yet.
Thanks

TITANForze3D, 2018-06-13:
Hello friends / Compiled via cygwin failed, downloaded from linux compiled files and used the script .bat / Update the firmware through win32 it turned out. there were additional commands - pwdmy, detectionmy, spi_flash_infomy (it does not work for some reason). Also, mf_ultralight_ev1 80b and 164b may appear. How to read information from Detection

TITANForze3D, 2018-06-13:
[image]

TITANForze3D, 2018-06-13:
ChameleonMiniLiveAndroid also writes that usb is not configured

TITANForze3D, 2018-06-17:
hello all,
thanks **ceres-c**
https://www.computerhope.com/issues/ch000549.htm - cygwin the compilation was successful
Mf_detection - could not catch a log
Terminal:

> --> rssimy?
> 101:OK WITH TEXT
>     0 mV
> --> spi_flash_infomy?
> 200:UNKNOWN COMMAND

So it should be?

ceres-c, 2018-06-17:
You can't run the flash info command, I am not sure why. Still, if your original problem is fixed, then you should be all set.

closed 2018-06-17 by TITANForze3D