iceman1001/ChameleonMini-rebooted #47: New device not working with Android NFC or card access system

RenwickCustomer, 2018-06-22:
Hi all,

I've got a new ChameleonMini Rebooted running the standard firmware, Chameleon-new-1.0.

Connecting to the device works fine, and it accepts commands without a problem. It also gets powered by readers without issue. However, I can't get my phone to recognise any emulated cards, nor a card reader to register it either.

Has anyone else come across this problem?

bogiton, 2018-07-12:
Have you tested the stock firmware or just this one? Someone mentioned (in issue #44) that the stock firmware worked fine (using the MF_DETECTION command) but not with this firmware. Is that your case too? What kind of emulation are you trying?

RenwickCustomer, 2018-07-12:
It's with the standard firmware that it came with, haven't flashed it at all. I've tried 1k and 4k at this point and got nothing. I also bought an ACR122U and tried it on that with no success either :(. Don't know if mine is just broken? It does light up when on a reader but isn't emulating anything by the looks of it.

I guess my current next steps are to try flashing this firmware? Having some issues getting the ATUMEGA  recognised by my machine using the drivers, but I probably haven't spent enough time trying! It's a bit taxing for someone who's not good at coding etc.

bogiton, 2018-07-12:
To be honest, I have only tried Chameleon on my proxmark, where it worked fine (probably because it is more fault tolerant than other readers). My only attempt on a real reader was by emulating UL-EV1, but with no success (as you can see in issue #34). But I believe that other users have successfully emulated MF Classic, using both the stock firmware and the new one. So, yeah, maybe you should try flashing and see what happens. What kind of software are you using with the ACR and the phone?

RenwickCustomer, 2018-07-15:
I agree - going to keep trying with flashing to see where I get! The phone is just the mifare classic tool and the ACR is Kali using libnfc tools. The ACR works with unlocked mifare 1K's, so it's definitely the chameleon i'm guessing!

iceman1001, 2019-03-08:
Any news on this one  ,  using latest firmware ?

slurdge, 2019-03-18:
With latest firmware, MCT on Android sees the ID but repeatedly loses the connection to the tag after that.

nicoschtein, 2019-06-10:
I'm having the same issue, MCT detects the key but looses connection while reading.

shinhub, 2019-08-14:
@RenwickCustomer @nicoschtein please try with firmware as of today.
If this issue is resolved, please close issue - and open new ones as needed.

iceman1001, 2019-08-18:
@RenwickCustomer  @nicoschtein  any news?  or is it time to close?

RenwickCustomer, 2019-08-18:
Hey both - I never was able to get the bootloader flash working, despite having the correct drivers installed. Very frustrating for me! It'll need extra time to investigate that unfortunately I don't have at the moment

I'll close it for you anyway, I can open a new one if required in future. Thank you @iceman1001 and @shinhub for your support through this regardless.

closed 2019-08-18 by RenwickCustomer