iceman1001/ChameleonMini-rebooted #49: Flashing issue

sh0eps, 2018-07-07:
Hi !

I spent my night to try to solve my issue but nothing worked... I wanted to upgrade my fresh Chameleon Rev E Rebooted firmware but I can't upload new files and the device is now stuck on bootloader mode.

First of all, to be able to compile, I had to remove the compilation flag Werror wish is in the makefile (was it on purpose ?). Then when I try to erase my device I have following output : 

`➜  ~ dfu-programmer atxmega32a4u erase --debug 100
     target: atxmega32a4u
    chip_id: 0x2fe4
  vendor_id: 0x03eb
    command: erase
      quiet: false
      debug: 100
device_type: XMEGA
------ command specific below ------
   validate: true

atmel.c:934: atmel_blank_check( 0x7ffeeb0048b0, 0x00000000, 0x00007FFF )
atmel.c:1041: atmel_select_memory_unit( 0x7ffeeb0048b0, 0 )
atmel.c:1073: Selecting flash memory unit.
Checking memory from 0x0 to 0x7FFF...
atmel.c:1100: atmel_select_page( 0x7ffeeb0048b0, 0 )
atmel.c:1114: Selecting page 0, address 0x0.
atmel.c:867: __atmel_blank_page_check( 0x7ffeeb0048b0, 0x00000000, 0x00007fff )
atmel.c:902: Region is NOT bank.
atmel.c:910:  First non-blank address in region is 0x7FF0.
atmel.c:981: Flash NOT blank beginning at 0x7FF0.
Not blank at 0x7FF1.
commands.c:87: erase 0x7FFF bytes.
atmel.c:444: atmel_erase_flash( 0x7ffeeb0048b0, 4 )
Erasing flash...  Success
atmel.c:494: CMD_ERASE status: Erase Done.
atmel.c:934: atmel_blank_check( 0x7ffeeb0048b0, 0x00000000, 0x00007FFF )
atmel.c:1041: atmel_select_memory_unit( 0x7ffeeb0048b0, 0 )
atmel.c:1073: Selecting flash memory unit.
Checking memory from 0x0 to 0x7FFF...
atmel.c:1100: atmel_select_page( 0x7ffeeb0048b0, 0 )
atmel.c:1114: Selecting page 0, address 0x0.
atmel.c:867: __atmel_blank_page_check( 0x7ffeeb0048b0, 0x00000000, 0x00007fff )
atmel.c:902: Region is NOT bank.
atmel.c:910:  First non-blank address in region is 0x7FF0.
atmel.c:981: Flash NOT blank beginning at 0x7FF0.
Not blank at 0x7FF1.`

I read other thread about it but I didn't find any answer. I tried on two macs and one Windows 10 device. dfu-programmer (0.7.2) output is always the same. and with the BOOT_LOADER_EXE I have the message "dfu-old-driver : no device present"

Thanks for your help

bogiton, 2018-07-12:
The only way to flash this currently is under Windows, as described in the Wiki.
As for the "no device present", please check issue #11

sh0eps, 2018-07-12:
@bogiton : thanks a lot.

to sum up for people in the same situation : 

-Compile as you can (I did it on Archlinux and OSX)
-In the flashtools folder, copy the hex and eep file AND the BOOT_LOADER_EXE
-Run flash.bat with device attached and driver ATXMEGA32A4U installed
-See the magic and enjoy your Chameleon :)

closed 2018-07-12 by sh0eps