iceman1001/ChameleonMini-rebooted #50: DFU driver (ATxmega32A4U) problems on WIN10 -

geiraage, 2018-07-11:
When I try to set the device from normal mode to DFU via GUI, a unknown decvice pops up in Device manager.

It I try to update manualy, and I point it to Chameleon Mini Rebooted GUI\ChameleonMini-rebooted-master\Drivers\DFU Driver (include subfolder), 
Its starts to install ATxmega32A4U, but it fails saying The system cannont find the file specified..

Its all files needed located in the repo?

bogiton, 2018-07-12:
Now that you mentioned it, I think there is a DLL missing from the repo. Try this and let me know if it works, so that I can update the repo with it.

[DFU Driver.zip](https://github.com/iceman1001/ChameleonMini-rebooted/files/2187782/DFU.Driver.zip)

sh0eps, 2018-07-12:
For your information the driver I used was in the dfu-programmer archive downloaded from original repo.

slurdge, 2018-09-01:
@bogiton The attached file works, so you can update the repo!

bogiton, 2018-09-01:
@slurdge Thanks for the confirmation!

closed 2018-09-01 by bogiton