iceman1001/ChameleonMini-rebooted #52: Getting ChameleonMini in DFU Mode

pitikay, 2018-08-28:
According to the documentation, I should be able to get the ChameleonMini into DFU mode by holding the black button while plugging it in. However, this is not working. I put the Chameleon into Sniffing Mode, and now it's not detected as an USB device anymore, and I need to reflash it somehow. Shorting PC3 and GND also does not work, maybe I'm doing the button press wrong?

slurdge, 2018-09-27:
I have the same problem, I believe some units don't behave properly.

NKT00, 2018-11-19:
I've just got one, and, well, which one even *is* the "right button"? One is red and one is black, mounted vertically, and the text is horizontal with the coil to the right.
So far it's thrown an error under Windows 10, which says it is a malfunctioning USB device, and under Ubuntu it doesn't do anything at all. 
Anyone got any ideas?

iceman1001, 2019-01-18:
I wonder which revision of Chameleon they are using...  Sounds like RevG

bogiton, 2019-01-18:
@iceman1001 Most likely :P

gtpy, 2019-01-18:
@iceman1001 not really- RevG doesn’t have colored buttons ;)
@pitikay @NKT00 Can someone upload a picture from your device?

iceman1001, 2019-01-18:
A **RevE rebooted** the buttons are RED and BLACK.    So their devices seem to be Rev E Rebooted.

Which firmware has sniffing mode?   Not RevE rebooted at least. 
_Shorting PC3 and GND _   ??  @bogiton  's first post says it all.

gtpy, 2019-01-18:
right :) skipped it

pitikay, 2019-01-28:
Thank you all for the replies, I attached a picture of my device - it's a chinese RevE Rebooted.
[image]

@bogiton There was a flag in the firmware for that specific mode, and well, I guess I played with it too much, without knowing what it'd do. Figured I could always boot it into DFU mode, but since it isn't detected by Windows or various distros anymore, and the button hold technique also doesn't work, I guess I'm out of luck with this one :)

slurdge, 2019-01-28:
@pitikay You have the same device as me. You can almost always salvage it by reprogramming it through the 4 little pins on the left of you CR2032 holder. You would need a special device though (or you can reprogram a small MCU to act as a debugger).

I had the same situation as you but it could be saved in the end.

iceman1001, 2019-01-28:
JTAG it...   thats the option that always work :)

shinhub, 2019-08-12:
@pitikay did you try to flash the original bootloader with original Windows flashing binary to the device, so that it becomes a dumb flashable brick you can upgrade with a new (working) firmware again?
As described in [here](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Getting-started#no-more-light-go-back-to-stock).

I then guess this issue could be closed @iceman1001 .

iceman1001, 2019-08-12:
Closed because of inactivity

closed 2019-08-12 by iceman1001