iceman1001/ChameleonMini-rebooted #53: Instructions for flashing the avr directly

slurdge, 2018-09-03:
While doing my experiments, something wrong happened and now my Chameleon is half-bricked.
Behavior:
 - If powered on by USB or battery+on button, the 5Th LED stays up red. Does not register at all on USB port.
 - If powered on and the black button is pressed, then register on the USB port but fails to deliver a USB device descriptor (so it's not a lack of drivers as with dfu etc.).
 - Does lights up quickly if presented near a NFC field.

So I think something has been corrupted inside the bootloader (I don't know how, as I never flashed it directly, it shouldn't be possible).
I would like to try to revive it but after searching a lot on the Internet, I did not find the specific steps and hardware needed for programming the atxmega32a4u directly. I know @iceman1001 does it, so it should be possible.

I found the 4 pins `3.3`, `CLK`, `DAT`, `GND` on the board, but I have absolutely no clue on how to use them nor the protocol used. From their location, I "guess" it's PDI ?

Maybe it would be useful to other people too.

iceman1001, 2018-09-03:
With a  ACRISP mkII you can recover.   It is on the wiki   [](https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Recovering-from-bootloader-and-code-overlap)

slurdge, 2018-09-03:
I understand that, but what about the pins?

iceman1001, 2018-09-03:
long time since I looked at it,  but normally thats where you hook up the MK-II

maxieds, 2018-09-04:
Has this link been mentioned:
https://lab401.com/products/chameleon-mini/software.zip. It saved my brick
of a Rebooted Chameleon on Windows 64-bit. Just wondering.

slurdge, 2018-09-04:
Hi @maxieds , thanks for the link. Unfortunately this would work only if the bootloader was working. It's analog to what you can find in the GitHub GUI repository.

bogiton, 2018-09-04:
@slurdge I think I was having problem flashing the bootloader when the battery was on the device. Try removing it and re-flashing using the BOOT_LOADER_EXE.exe. It may solve your issue without an external programmer.

bogiton, 2018-09-04:
Also, check this out: https://lab401.com/products/avrisp-mkii
This is the MK2 with a Chameleon shield that seems to be using SPI over 3 pins (MISO, SCK, RST). So, maybe you connect the MISO pin of your external programmer to the DAT pin (?) of the Chameleon.
There is also this adapter from the same store: https://lab401.com/products/chameleon-mini-avrisp-adaptor

slurdge, 2018-09-04:
@bogiton thanks. I already removed the battery. I'm pretty sure (like, 99%) that if the dfu does not come up in device manager or lsusb there's not hope unless I flash directly.

Unfortunately I do not possess a mkii. I'll order one. The connection should be straightforward even without a special adapter since there is so few pins, I'll just to figure out which one to use.

I'm pretty stumbled as why the problems appears in the first place, as I dfu-programmer can't overwrite the bootloader. I'm leaning toward a flash problem during the numerous plug-in plug-out for my tests.

slurdge, 2018-09-27:
Success! I received a  AVRISP mkII and I could reprogram the Chameleon mini.
Maybe you can include in the wiki the pin connection:

AVR   | Chameleon
----- | -------------
VCC  | VCC
GND | GND
MISO| DAT
RST(SCK)|CLK

With that, you can reprogram the Chameleon mini. Worth to be noted, mine was in such a bad state that I had to erase it completely as the lock bits were damaged.

I can't edit the wiki but I think is the above info is included we can close this issue easily.

iceman1001, 2018-12-17:
https://github.com/iceman1001/ChameleonMini-rebooted/wiki/AVRISP-mkII--Pin-layout

closed 2018-12-17 by iceman1001