iceman1001/ChameleonMini-rebooted #55: Automatic builds of the firmware

jlirochon, 2018-09-11:
Hey @iceman1001,

Is it legally possible to publish binaries of the firmware ?

If it's legally possible and you are OK, I can help with the automation (automatic builds upon GitHub push).

jlirochon, 2018-09-11:
Great, go for it ! You can flag this with `enhancement` :)

iceman1001, 2018-09-11:
Just make it, bro!  I will merge it

iceman1001, 2019-02-22:
What happend with automatic builds!?!  @jlirochon

slurdge, 2019-02-22:
I had a look and ended up with this travis.yml file that does work for me.

https://github.com/slurdge/ChameleonMini-rebooted/blob/master/.travis.yml

iceman1001, 2019-02-22:
PR?

slurdge, 2019-02-22:
Sure, but I think you and/or @bogiton would need to setup the appropriate travis-ci key to be able to push here. Also, do you prefer it squashed or as is ? I did quite a lot of trial/errors :-p

iceman1001, 2019-02-22:
squash please ;)
I think I already have a travis-ci key,  from the proxmark repo.    Shouldn't be too hard to set it up.
Or did you do that branch "travis-ci" solution?

slurdge, 2019-02-22:
Nah I didn't do the branch solution. I'm not completely fluent in how the travis-ci keys work, so you have need some adjustments (strangely I have double builds done on my fork).

iceman1001, 2019-02-22:
turns out I have travis-ci already setup prepped already for this repo.   Lets see how to get it to work

iceman1001, 2019-02-22:
I think we have success?!?!

[image]

iceman1001, 2019-02-22:
I can see some problems with this in the long run.
Alot of releases and strange naming convention.

closed 2019-02-22 by iceman1001