iceman1001/ChameleonMini-rebooted #57: Mifare Classic Clone + Reader Attack

X0x1RG9f, 2018-09-14:
Good Afternoon,

I just bought a Proxmark3 RDV4 and the Chameleon Mini RevE Rebooted. I'm starting to play with both, but  I'm a little stopped in my enthusiasm with very simple things on Chameleon. I don't know if it could be a chameleon hardware issue or simply one thing I did not understood, so I will try to explain my best with what I know !

### 1st Issue : Cloning a Tag
I have a simple tag for entering my building which is a MIFARE CLASSIC 1K, as show below :
```
proxmark3> hf search
hf search

 UID : be ** ** **
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1
proprietary non iso14443-4 card found, RATS not supported
Chinese magic backdoor commands (GEN 1a) detected
Prng detection: WEAK

Valid ISO14443A Tag Found - Quiting Search
```
Copying that tag on a simple Magic Chinese Card works fine with the bellow commands, and I can enter the building : 

```
hf mf mifare
hf mf nested 1 0 A ************ d   (*** being previous found key)
hf mf dump
hf mf restore 1 (on a BLANK card)
hf mf csetblk 0 xxx...xxx (I set it manually because for unknown reason it cannot be written with restore)
```
This is working fine, and I would like to reproduce exactly the same behavior with the Chameleon. However, after some days trying and trying, whatever GUI I use, its ends in a big failure :(.

My steps in the GUI are the followings : 
1. Connect to the Chameleon (Firmware Chameleon-new-1.0) in the GUI
2. In "Operation" tab, select "Slot 1", and then select MF_CLASSIC_1K
3. I then click on "Upload Dump" and upload the previously created dumpdata.bin with proxmark
4. UID is then replaced by my dump
5. I click on "Apply" 

At the contrary of Magic Chinese Card, this is not working at all with the building reader(s). When I approach the Chameleon from the reader, red light of the TAG1 is ON (sometimes fixed, sometimes blinking, can't explain). Readers, on their sides, sometimes pop a red light, or most often nothing, like if nothing was read from Chameleon. 

Also, when I try to read the Chameleon back with the Proxmark, I sometimes have some reading errors : 

```
proxmark3> hf mf dump
hf mf dump
|-----------------------------------------|
|------ Reading sector access bits...-----|
|-----------------------------------------|
[...]
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
#db# Cmd Error: card timeout. len: 0
#db# Read block error
#db# READ BLOCK FINISHED
#db# Can't select card
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
|-----------------------------------------|
|----- Dumping all blocks to file... -----|
|-----------------------------------------|
[...]
Successfully read block  2 of sector  7.
#db# READ BLOCK FINISHED
Successfully read block  3 of sector  7.
#db# READ BLOCK FINISHED
Successfully read block  1 of sector  8.
#db# Can't select card
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
Successfully read block  2 of sector  8.
#db# READ BLOCK FINISHED
[...]
Successfully read block  0 of sector 10.
#db# Authentication failed. Card timeout.
#db# Auth error
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
Successfully read block  1 of sector 10.
```

Am I missing something here ? I must admin that I have no idea of what can I be doing wrong, any help could be appreciated on that subject. 

### 2nd Issue : Reader Attack
I also tried the reader attack presented in that video (https://www.youtube.com/watch?v=1VpXC3-eKhc), configuring the "Slot 2" with MF_DETECTION and inserting a valid card UID.

When I approach from the readers again, same behaviour, sometime fixed red light, sometimes blinking.

I then come back on the GUI and tries a mfkey32 on the "Slot 2", but again, nothings happens and absolutely nothings appear on the screen. 

Any idea on that behaviors also ? 

I'm a beginner on these subjects, and your help would be really appreciated. I don't have any problems with the Proxmark, but actually I cannot do anything with the Chameleon ! 

Thanks for your attention and your help !

X0x1RG9f, 2018-11-28:
Hello,

Any news on the subject ? Has someone experienced the same issues ?

Regards,

ceres-c, 2018-11-30:
I have not yet used my chameleon to emulate Mifare cards, so I can't help you in this particular matter
On the other side, I'd suggest you to try and use the terminal instead of the GUI, it might yield different results.

yaxax, 2019-03-13:
Hi  can some one help me out.I don't understand the output off the chameleon mini reve when I pull the device off the reader and then connected it back to my laptop.I see traces but don't know how to go further .

yaxax, 2019-04-14:
How can i upload  the dump off the prox3 to the chameleon mini reve rebooted?I whant the dump of the prox3 in my chameleon mini rebooted.

yaxax, 2019-05-11:
I use the Chameleon mini reve rebooted  to read from reader and then crack it to find a key whit mfkey32 but when I upload the dump and replay it the door still lock just a red light bling what im doing wrong? And the reader attack can i make a clone card whitout accessing to the rfid card?

yaxax, 2019-05-11:
But when i use the proxmark3 and dump the card to dumpdata.bin And then upload the dumpdata.bin in to the chameleon mini reve rebooted the door open.

bogiton, 2019-05-17:
@yaxax If you can emulate using Chameleon, after uploading a valid dump, then I see no issue here. You can't read a tag using the Chameleon, it can only be used for emulation.

yaxax, 2019-05-17:
Thanks for your answer.I know that the chameleon can't read cards,but is about the reader attack when I upload the dump.bin 4096 bytes off the chameleon off the reader attack and replay it back the reader then i get red light not green And allso i see in the dump management that when I do reader attackt it's always the same info when I upload the chameleon bin.only the uid not.So I can't do a replay back to the reader whit the dump.bin. 4096bytes?are do I must make the 4096 to a 1024 bytes to do a replay back to the reader?

bogiton, 2019-05-17:
The MF_DETECTION only uses the UID. It is irrelevant if you upload a dump or not using that method. Also, as far as I know, it only supports MFC 1K. So, you can't do a "reader attack" against a reader that expects MFC 4K.

closed 2019-05-17 by iceman1001

yaxax, 2019-05-18:
Thanks for the answer.I know that chameleon only use uid and I allso know that you can't use a 1k if the reader only supports 4k.But I put a 1k uid in mf_detect the reader only supports 1k but in the  upload is 4096 bytes so if I put the same data but in a 1024bytes does the reader accept this?