iceman1001/ChameleonMini-rebooted #59: is there a GUI for android which like the one on PC?

K7-Lenovo, 2018-10-30:
now i use ChameleonMini live debugger, but i think this app is hard to learn to use than windows GUI by iceman. I still did not know how to upload a dump file to my devices. i think iceman GUI for windows is very great, no useless button and very suitable for ChameleonMini-rebooted. ChameleonMini live debugger have so many button but my device is ChameleonMini-rebooted, lots of them is not support, looks very confused.

K7-Lenovo, 2018-10-30:
do anybody have suggest? i think i can use my ChameleonMini very good on PC, but on phone i can only use it by terminal command. but i can not upload dump to it by command, is anyone know how to upload card by using a android device

iceman1001, 2018-10-30:
Sorry,  I have no idea about android GUI for chameleon mini.

maxieds, 2018-10-30:
Yes, see https://github.com/maxieds/ChameleonMiniLiveDebugger. There is a
link on the GitHub page to the app on play store, or it can be installed by
the apks posted on the releases page. Works with both rev g and rev e
boards.

maxieds, 2018-10-30:
Sorry, just responded to your message based on Iceman's last post. What's
hard to use about the app? If you know the commands from the reference
(i.e., what you are typing at a serial terminal), then you just press the
corresponding button in the GUI. I'd be interested in making it more user
friendly in future versions.

iceman1001, 2018-10-30:
:)  I had to edit my post to make it clear,  since I do know of yours @maxieds . I just don't know of other ones.

The idea of a one button gui is a strong one.

iceman1001, 2018-10-30:
maybe a gui which solves some / one problem..  not be the complete gui...

maxieds, 2018-10-30:
So do you think a separate simpler GUI just for the rev e boards is a good
idea? There's also this library which I could reuse:
https://github.com/maxieds/ChameleonMiniUSBInterface.

bogiton, 2018-10-30:
I am willing to help on making one.

iceman1001, 2018-10-30:
A seperate simpler GUI sounds like a very good idea.  You might just make it so simple that it would even work for both RevE, RevG...

One day maybe we even get a _RevG Rebooted_  revision :)

closed 2018-10-31 by iceman1001

K7-Lenovo, 2018-11-01:
> Yes, see https://github.com/maxieds/ChameleonMiniLiveDebugger. There is a link on the GitHub page to the app on play store, or it can be installed by the apks posted on the releases page. Works with both rev g and rev e boards.
> […](#)

ok, thanks. i will find some useful wiki in that github page and try ChameleonMiniLiveDebugger again. also, do you have any idea about how to upload card by using a android device

K7-Lenovo, 2018-11-01:
> A seperate simpler GUI sounds like a very good idea. You might just make it so simple that it would even work for both RevE, RevG...
> 
> One day maybe we even get a _RevG Rebooted_ revision :)

thanks iceman, i think make a GUI only suit for our ChameleonMini-rebooted is a good idea, it can be a very easy and useful App like PC GUI. Also, can we find a solution for how to upload card by using a android device now? i found i cant do this by using serial terminal  command.

iceman1001, 2018-11-01:
Speaking of which, but doesn't the Team Walrus project work on Chameleon minis aswell?

K7-Lenovo, 2018-11-01:
> Speaking of which, but doesn't the Team Walrus project work on Chameleon minis aswell?

no, walrus isn't support ChameleonMini-rebooted, also i found pm3 rdv40 can be find by walrus but i got no respond when i put card on my pm3, nothing happen on my screen and pm3, still "waiting for cad"

iceman1001, 2018-11-01:
I think the Walrus project has the intention to support Chameleon..  Even RevE Rebooted..
They have a slack channel for easy chats.    Why not ask a question there?

K7-Lenovo, 2018-11-01:
> I think the Walrus project has the intention to support Chameleon.. Even RevE Rebooted..
> They have a slack channel for easy chats. Why not ask a question there?

ok, what is that channel, could you please gave me a link? i had start a issues in walrus github about PM3 RDV40([here](https://github.com/TeamWalrus/Walrus/issues/110)), but no reply until now.

iceman1001, 2018-11-01:
If you read the first page on their repo,  you will find all your answers.   (https://github.com/TeamWalrus/Walrus)

K7-Lenovo, 2018-11-02:
> If you read the first page on their repo, you will find all your answers. (https://github.com/TeamWalrus/Walrus)

ok ,got it, thanks. could you leave your email here

K7-Lenovo, 2018-11-02:
> my email isnt very hard find on github....

yeah, i see that in your profile, i am ask because i an not sure are you still use that email.