iceman1001/ChameleonMini-rebooted #61: Problem to use/handle the Chameleon mini RevE Rebooted

peseta3, 2018-12-07:
Hallo Together,
@iceman1001 

last Year I get an Chameleon Mini RevE Rebooted from Lab401.com
I had so my Problems with this.
[(https://github.com/iceman1001/ChameleonMini-rebooted/issues/9)](url)

Now I want to try it again and hope it will now work.
I use Windows10 and use some GUI`s, the old from last Year with calc/fast calc Bottum, from Iceman, from LAB401and TeraTerm.

Windows DeviceManager:  Anschlüsse(COM&LPT)---Serielles USB-Gerät (COM3)

[image]

They all say`s that it is connected.
The firmeware is Chameleon-new-1.0

[image]

[image]

Is this the correct Firmware?

I want to snif a Cardreader ACR122u with the Chameleon, but nothing happens.
I think normaly the Chameleon LED should be blinking shortly, but the LED is on th hole time if I but it on the Cardreader.
I use a battery from Chameleon too.
Before I use it I connected with a PC Windows10 with GUI.
Firmware is Chameleon-new-1.0
Slot1   MF_DETECTION  SWITCHCARD  APPLY  SET ACTIVE

I use this viedeo for Tutorial
[https://lab401.com/blogs/academy/chameleon-mini-mifare-cracking-via-the-reader-attack](url)

Are you  any Ideas? Whats wrong?

peseta3, 2018-12-17:
No one any Idea?
If I use the GUI the serial Part, and I type  RSSIMY?
I get this:

`--> RSSIMY?
101:OK WITH TEXT
    0 mV`

Is it normal, if I have a Tag/ Mifare_1k near the Chameleon?

iceman1001, 2018-12-17:
the chameleon-new-1.0 firmware string is a sign your firmware is of the  RevE rebooted which as a side note  has changed since we got it open-sourced again.

One of the eight slot leds should be on, indicating which slot you are currently using.
If you download latest GUI,  you will see it clearly marked with green borders.

The obvious question is if you have configed the active slot correct. So that it has something to answer your ACR122u reader when it issues a command to your chameleon.   I don't know how you are control your aCR122u, since you didn't say  but I assume you have some software where you can tell it to identify tag.

peseta3, 2018-12-17:
Ah Hello Iceman,
thank you for your answere.
The ACR122u reader use the original ACR122U Windows Software.
And the output is the correct Data from Slot 1.
But how can I use The Chameleon stand allown to pick up Data from Card or Reader?
And use them with PC and USB Cabel as a Reader?
I didn`t get any Data from e.g. Mifare classik 1k.

iceman1001, 2018-12-17:
I think you have missunderstood the chameleon device.   It can simulate a tag, meaning you need to upload a tag memory dump onto a tag slot.   Ie,  1k mifare classic dump.

Then you present the chameleon to the reader,  which can do stuff.  
Then you hook up the chameleon with the GUI, to download the memory again.

You can now analyse what the reader did to the card.
You can also use the mode  MF_DETECT,  and present to a reader,  which tries to authenticate to the chameleon.   afterwards, you hook up the chameleon to the device and run "mfkey32 attack",  in order to potentially get the key out from those reader authentication attempts.

peseta3, 2018-12-17:
ah, yes I understand.
 but with this

> You can now analyse what the reader did to the card.
You can also use the mode MF_DETECT, and present to a reader, which tries to authenticate to the chameleon. afterwards, you hook up the chameleon to the device and run "mfkey32 attack", in order to potentially get the key out from those reader authentication attempts.

I want to try this, but it doesn`t work.
I think Slot 1 LED should be blink if I put it to the ACR122u.
But the LED is only on the hole Time.
And if I make mfkey32 Attack. I get.

`mfkey32 attack failed, no keys found`

peseta3, 2018-12-17:
With the ACR122u, I do nothing, he is only conected with USB and the Software.
He is in read modus, to read all cards and coneced them or not.

iceman1001, 2018-12-17:
...but what does the  ACR122u reader read?   only uid?   or does it try to read the whole memory from card?

peseta3, 2018-12-17:
If i click on read "get data" he read the UID.
If I put only the card on the reader the green led is on. and he accept the card

iceman1001, 2018-12-17:
So the chameleon works and this is not a firmware issue.

Somewhere in you AC122u software, you will need to tell it which keys to use etc in order to be able to authenticate against the card.  Without it,  it can't read the memory.

peseta3, 2018-12-17:
Yes sure you are right thats not an Firmware Issue, with your Work is all ok and great.

ok I understand.
But how is it to use the Chameleon as reader and Read a Card Standalone?
Can you tell me about this a little bit.
Yes I know its not your Problem, but I have the Chameleon one Year an I can not realy work with them without some Informations for it. I read realy many of Documents, but nothing about handling it.

peseta3, 2018-12-17:
Hmm 
but Lab401 wrote me that he can Read

> The device has specific read modes, and can also be written to.

But he should be comunicate with the e.g. Door acces reader, I saw it in the Video from LAB401.