iceman1001/ChameleonMini-rebooted #62: Bricked after failed firmware upgrade

jwgn, 2018-12-14:
Hi,

I tried to perform a firmware upgrade, but it failed:

Creating the EEPROM binary...
Write done!

Creating the Flash binary...
B:\Downloads\avr-objcopy.exe:B:\Downloads\Chameleon-Mini.hex: File format not recognized
Not find fileB:\Downloads\flash.bin konnte nicht gefunden werden

Flashing the files onto the "Chameleon-Mini Rev-E Rebooted"...
old_driver_bootloader
Erasing flash... Success
Checking memory from 0x0 to 0x6FFF... Empty.
0% 100% Programming 0x80 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>] Success
0% 100% Reading 0x400 bytes...
not_find_file

If there are no errors above, flashing the firmware to your "Chameleon-Mini Rev-E Rebooted" should be finished now. Enjoy!

Now I am unable to restore the Chameleon. It starts directly into bootloader mode when I connect it via USB (no LED flashing, shows up in device manager as ATxmega32A4U) but it is no longer being recognized by the flashing tools:

B:\Downloads>BOOT_LOADER_EXE.exe
old_driver_bootloader
dfu-old-driver: no device present.

Any Idea on how to get it back to work?

slurdge, 2018-12-17:
Look at issue #53

iceman1001, 2018-12-22:
is this issue resolved?

closed 2018-12-22 by iceman1001

jwgn, 2019-01-08:
I think I need a bit more assistance here. This is where I ended up now:

[image]

Flashing works, but I have no clue if I'm doing sane things. The Chameleon is not being recognized at all after flashing. It only glows red when I connect it via USB. Nothing in Windows Device Manager or in Linux dmesg. What am I doing wrong? Its hard to find suitable all-in-one instructions for recovering a bricked device.