iceman1001/ChameleonMini-rebooted #63: Flashing new device

wk-it, 2019-01-04:
Hello

Today I got my Rev E rebooted.

As suggested I tried to flash the firmware as the installed is just for shipping!?

I can connect to the Rev E on Mac, Linux and even on Windows.
I can get it in to Bootloader Mode.

Thats where I'm Stuck.

In Windows it is shown as Atmega 32 Device (When in Bootloader mode). But the Flash.bat always gives error: "no device present"

On Linux (in bootloader mode) it is recognized as VendorID 03eb and Product 2fe4
AVRdude tells me: "Error reading XMEGA config information."

Tried the guides from the wiki here.

Built the FW on Linux, copied it to Windows but it seems that I have missed something.

The device itself should be ok, after replugging I can connect through the gui tool in Windows and also with minicom in Linux

Hope you have a hint, 
thanks!

bogiton, 2019-01-04:
Hey there.
Have you tried installing the DFU drivers while in bootloader mode?
Have you got all the required files in the same dir as flash.bat?
Are you sure that no other software is accessing the Chameleon when you run the flash.bat?

wk-it, 2019-01-04:
I installed them and then connected. 
Eben after reboot no luck. 

You mean connect it in bootloader mode and then install the driver?
Which one do you use? from this repo?

All files are present. Creating the bins also work. Just stuck at dfu with „no device present“

Yep, nothing else running

bogiton, 2019-01-04:
Yes, keep pressing the white button while connecting the device to enter the bootloader mode and then try to install the DFU drivers from this repo.

bogiton, 2019-01-05:
Try to install the libusb-win32 driver with zadig (https://zadig.akeo.ie/) instead. Hint for zadig: Check the "List All Devices" option from the "Options" menu to view the Chameleon.

wk-it, 2019-01-06:
I made it!
I used the Windows Power Shell instead of the old "command line" then it worked.
Nothing else changed.

But I think I misunderstood the Features Page from lab401.
I thought it is possible to read tags.(Mifare classic)
As I think now, it is only possible to upload dumps to the RevE and emulate the cards.

Am I wrong?
Should I bought the Chameleon mini instead!?

PeterHuewe, 2019-04-16:
I have exactly the same problem - I cannot flash from windows or linux with same symptoms.
Using powershell was not the issue.