iceman1001/ChameleonMini-rebooted #64: Software and MF0ICU1 questions

olaf1234, 2019-01-17:
Ty for your anwser @bogiton

One question lead to another:

Some context : When i'm reading my MF0ICU1 emulation with my NFC taginfo app, the card UIDS,BBCS, OTP, LOCKBITs, DATA, manufacturer, RF technology, ATQA and SAK are identical to the real one. I uploaded on the chameleon a .bin file of my dump. Althought, even if the access conditions are read correctly (i got 3 read only pages), I can still modify them (include OTP and lock bits) with my MIFARE++ ultralight app. When i present my chamelon emulation to the real reader/writer : Incompatible card.

Questions: Does the state of my chameleon could be the problem(ex : IDLE, Halt)? , Does the real access conditions ( not the ones read) is noticed by the real reader/writer. Could it be something else? If those questions are valid interrogations(or not lol), can anyone give me some hints to resolve this? I'm pretty new to this.

bogiton, 2019-01-18:
This could be related to the Chameleon timings, as mentioned also in issue #34.
The Chameleon should be able to switch its state according to the queries from the reader. One thing you could try is to sniff the communication between the Chameleon and the real reader with another device, like the Proxmark3. That way, if it's not the timings, you could probably identify the reason it rejects your emulated tag (or if the Chameleon itself responds badly to a specific reader command). I would be very interested in the results if you do this research.

iceman1001, 2019-01-18:
Don't think Ultralight support is that bad at all.  No crypto involved.

olaf1234, 2019-01-18:
@bogiton 
@iceman1001 
Ty for your responses.

After some aditionnal reading, thought it would be important to mention im running a default rev-e rebooted 1.0 firmware. Flashing the newest might solve the problem?

Chameleon timings: What exactly are you referring to? Can I calibrate that?

iceman1001, 2019-01-19:
Try flashing to the latest firmware from this repo?

olaf1234, 2019-04-27:
Hello, 
I had to take a break. I had to many side projects going on. I flashed your 1.3 firmware. Still ain't reacting.  I started to read about proxmark3 and decided to order one. I'll keep you guys updated.

olaf1234, 2019-05-01:
@bogiton
@iceman1001

I have received the proxmark3. I am using: 
 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
  [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-980-gbacf8aff 2018-08-04 13:36:38
      os: iceman/master/ice_v3.1.0-980-gbacf8aff 2018-08-04 13:36:42

Correct me if i'm wrong, in order to log all traffic between the real RFID readers and  my chameleon mini 1.3 iceman firmware. 1) I need to run the command ''hf snoop'' then place the antenna in between the reader and the chamelon. 2) I press the hardware button to stop logging. 3) I run the command ''hf 14a list'' so I can read the log.

I just wanna make sure i'm correct because bringing all that hardware will look a bit suspicious, I want to minimize my attempts lol.

iceman1001, 2019-05-02:
this is a Chameleon-Mini repo,  questions about PM3 is best asked over at the proxmark3 forum,   and its a good source of information to learn how to use your new proxmark3 device.  LIke the PM3 wiki,  or blogposts, or youtube videos.

shinhub, 2019-08-14:
@olaf1234 try with firmware compiled as of today if need still be.
@iceman1001 shall we close?

closed 2019-08-14 by iceman1001