iceman1001/ChameleonMini-rebooted #65: Black Key Switch button stopped working

dealaddict, 2019-02-06:
I successfully flashed my chameleon mini with the firmware here and my chameleon mini detects fine in rebooted gui. However the switch button is not working and it is not able to switch cards. This only happens on battery mode. While it is connected to the computer, the switch and lights work, but once disconnected on usb the button no longer works.

Appreciate your help.

iceman1001, 2019-02-06:
try pressing the red key and keep it pressed,   then press black key to swap between slots...

dealaddict, 2019-02-06:
Tried it but it's not doing anything. Only the power button lights up but doesn't switch key.

Can you think of anything why this is happening? I tried reverting back to stock with those two stock files, it flashed successfully and went back to stock but issue remains. 

It's kinda weird it that those buttons work fine when it is connected with a cable to the pc or to USB otg phone.

bogiton, 2019-02-06:
Are you sure you have set the button action to SWITCHCARD for that slot?
Only when the action is SWITCHCARD, you can switch tags using the black button.
In any case, in order to use the black button, while on battery, you have to keep pressing the power button, like iceman said, to power the device first.

dealaddict, 2019-02-06:
Yes all my slots are set to Switchcard option. Anyway strangely enough now I can switch them by pressing the power button and then the switch button. Just curious was it always like this? I am pretty sure I was able to switch cards by just pressing the black button before. 

But hey atleast it works, I guess I have a defective piece because my battery holder slot is pretty flimsy and it feels like it could come off any moment, even the case doesnt fit properly near the micro usb slot side.

closed 2019-02-06 by dealaddict