iceman1001/ChameleonMini-rebooted #66: Supported dump files to RebootedGUI

dealaddict, 2019-02-06:
Is there any android app that makes dump files supported by RebootedGUI? I tried MIFARE++ Ultralight but it makes dump files in .txt format and despite playing around with them it wont upload properly. I got excited about the Ikarus tool but its for Classic only and I need something that supports Ultralight dumps.

FYI I already bought the ACR122U but its on its way and may take a month so in the mean time need something to play with.

Thanks

bogiton, 2019-02-06:
Maybe we could provide a convert tool, if none of the existing ones is suitable. Can you share the txt?

iceman1001, 2019-02-06:
One of my ideas years back when I saw MCT app the first time,   'hey,  there should be a MUL app...'   and then everything else got it the way but yes I agree.

iceman1001, 2019-02-06:
However its not a question for the firmware of Rebooted..  So I will close this,  you can conside opening it under RebootedGUI repo instead.

closed 2019-02-06 by iceman1001

dealaddict, 2019-02-06:
> Maybe we could provide a convert tool, if none of the existing ones is suitable. Can you share the txt?

I just posted them in Rebootedgui repo