iceman1001/ChameleonMini-rebooted #68: My Chameleon is bricked

dealaddict, 2019-02-11:
It was working fine till this morning but now it's not connecting anymore. There is no lights with or without the battery and even on usb connection. It gets detected in Chameleon REBOOTEDGUI for like 3 seconds with a success message and then disconnects.

I was running the latest firmware. 

Is it possible to fix my device?

bogiton, 2019-02-12:
This is weird. It just stopped working? The signs are pretty bad to be honest.
Maybe you can try the guide shown here: https://github.com/iceman1001/ChameleonMini-rebooted/wiki/Recovering-from-bootloader-and-code-overlap

dealaddict, 2019-02-12:
Yeah I uploaded all my dumps and it was working fine. Left it for a few hours and suddenly it stopped working. It definitely detects and connects for like 3 seconds before it disconnects. That time is not enough for me to put it in DFU and flash it.

Is there no other way than using the atmel connector? It's almost the same price as a new chameleon lol @bogiton

bogiton, 2019-02-12:
I'm sure you can find very cheap clones of the avrisp mkii from china. I haven't personally tested any of them though.

iceman1001, 2019-02-13:
unplug the battery?

dealaddict, 2019-02-13:
Update: Unplugging the battery didn't help but I manage to unbrick my device with a bit of stroke of luck and Iceman's old software. I realised that the unit was booting for 2 seconds before it was bricking so I just timed about right and managed to put the unit in DFU mode by using his older version of GUI and then quickly running the bootloader file. Needed few times to get the timing right (connecting cable, putting into dfu, opening the bootloader program etc) and it eventually managed to reflash it just in the nick of time. And voila! The unit booted up and powered on with lights and I was able to bring it back to life. Thanks @iceman1001  @bogiton

closed 2019-02-13 by dealaddict

iceman1001, 2019-02-13:
Great!
even if your way is doable I recommend using the **AVRisp-MKII**