iceman1001/ChameleonMini-rebooted #69: I can't seem to emulate mifare ultralight reliably

ph553, 2019-02-11:
Hello,
i have tried using the chameleon for mifare ultralight emulation, but i can't seem to emulate it reliably
When using my acs122u reader, the green light on the reader keeps strobing(green then off) and it gets detected on windows for a split second before disconnecting, and reconnecting again and again.
I have tried to read the chameleon mini from another reader(the one that my emulated card is used on) and it works sometimes, and some other times it doesn't.

I have noticed that, when i upload a dump on the chameleon and immediatly download it back, it adds 12 bytes of "1A" data, making the mifare ultralight card 64 bytes + 12 bytes
Is it a problem with my chameleon or my firmware?

bogiton, 2019-02-12:
Those 12 bytes appended at the end are probably the counters.
As you can see in issue #34 it is known not to work properly, at least for Ultralight EV1.
Other users also seem to have problems with Ultralight (see issue #64).

ph553, 2019-02-12:
Thanks for the answer
Do you know if it's fixable with a firmware update or is it a hardware problem?

bogiton, 2019-02-12:
Well, I really hope it can be fixed firmware-wise. The hardware most likely is capable of emulating UL.
If we could detect why sometimes works and some others doesn't, then we might be able to get to a solution. Sniffing both communications (valid and invalid) between the reader and the Chameleon is definitely a first good step.

ph553, 2019-02-12:
That was about to be my next question;
Can i sniff communications with the chameleon? When i put it in mf_detection mode i can't change the uid length to be a valid UC uid
Is there a sniffing mode i'm not aware of? If yes, how do i put it in sniffing mode and how do i read the comms log?
I might be able to post the log of a working comm vs a non-working one, or even try to fix it myself if it helps

bogiton, 2019-02-12:
No, you can't sniff using the Chameleon. You need another device, like a proxmark.
But, in any case, you would need a separate device, in order to sniff the communication between the Chameleon and the reader. Especially, if we wanted (and we do) to also log the timings of the bi-directional commands.

ph553, 2019-02-12:
Alright thanks for your time! Appreciate it

iceman1001, 2019-02-13:
time to close?

closed 2019-02-13 by ph553