iceman1001/ChameleonMini-rebooted #70: Few questions about ChameleonMini Rebooted

dealaddict, 2019-02-13:
I need this device as a daily driver for work purposes so wondering in real world how durable it is? Also I feel the units are poorly designed, there is a possibility they can brick themselves and even the battery compartment and buttons feel flimsy maybe its just my unit or this is the case for everyone?

Also does anyone know what is different in Piswords Chameleon? I found this online and it said it is running a different code. I haven't bought it but just curious.

iceman1001, 2019-02-13:
Wut??   Flimsy buttons?  poorly design units?    
Did you buy a original or a china clone of the RevE rebooted?   (I know,  rev rebooted is a chinese remake of the RevE... but there exists clones of that remake)     My RevE doesn't brick itself nor is the buttons flimsy. The battery fits quite well and doesn't move.  I had it for the better part of two years. 

When you say flimsy buttons what do you mean then?

dealaddict, 2019-02-13:
Thanks for replying. By flimsy I meant the usb slot feels loose, I just has to connect the cable right or else it wont detect, the same with buttons it doesn't feel fully secure and I feel eventually it may break or fall off (talking about the tag button) and the power button doesn't work well with case on so I have to really press down on the case to press it as it stays slightly open. 
Same issues with the battery compartment, one side of the slot is popping off and I may have to solder it properly. And then yesterday it literally bricked itself and wouldn't boot up....

I guess its just my unit which is acting up. I will look into a replacement.

bogiton, 2019-02-13:
I find my device very well made. No flimsiness to it :P I got mine from lab401.
I wonder why Piswords claims that it is their design... You can see clearly on their images that is says: "Proxgrind", same as the original RevE rebooted...

iceman1001, 2019-02-13:
I missed that  **Piswords Chameleon**  part.   Its clearly a Proxgrind build.
I only use my device with the case.  I never use the bare pcb.  All your issues sounds its a bad soldered unit.  If this is one of these chinese clones of proxgrind then you got what you paid for.
My unit is from proxgrind,  and yes,  lab401 sells only proxgrind units. 

So either @dealaddict 's unit is a chinese clone or you really got a bad unit.  Either case I agree that you should RMA it and ask for a replacement device.    The bad soldering could also be the cause for the bricking.

nikmel420, 2019-02-14:
its made very well. theres no software to nstall on it. for 3 days ive gone throgh so many mazes . tis bin goes here. do this now . wait on windows. the istruction mk no sence and sndyou everwere but a solution.theres like 20 diferent ufinished versions and not one done. you have windows and linux in the same spot' ends up messing up both pcs. you made a gui before the firmware. dont buy this unless u can writ your own firmware.

iceman1001, 2019-02-14:
@nikmel420  I hear you frustration but sorry to say,  we didn't "create" this product or gui to start with.
The volunteers of this repo has spend hours trying to figure out how to update, liberate and make easy gui help ppl to get a nice easy experience.   You send to be confused about that.  We spent months so ppl should get a possible way of liberate their device.  We spent months doing improving a GUI to make it really easy and nice to work with. 

If you don't like our free work feel free to keep your opinions to yourself and use other tools & software.

dealaddict, 2019-02-14:
In my personal opinion the efforts of people like Iceman and Bogitan have been invaluable to me. Because of their software and helpful advices I am able to do things I never dreamt were even possible. If it wasnt for them we would be dealing with some chinese softwares and non existent supports. Their software has bailed my ass more times than I could count. I don't know A of typing any commands of unix so having everything readilly made to me with one click of a button is godsend.

closed 2019-02-14 by dealaddict