iceman1001/ChameleonMini-rebooted #71: firmware

nikmel420, 2019-02-14:
can u just make a simple way to upgrade the firmware. and give instructions fully begining to end. dont give half linux then switch to windows intructions. its a clusterfuck! theres old stuff with new clean it up . make it logical. please how do i update the firmware that came with device.

iceman1001, 2019-02-14:
I agree that a cleanup of the information gathered would be in place.  

Feel free to contribute.

closed 2019-02-14 by iceman1001

dealaddict, 2019-02-14:
For me it was pretty easy, I just installed the two drivers, removed some lib file from the folder, paste the two firmware files in the same folder and used the rebootedgui to put my chameleon into bootloader and ran the program. It was pretty easy for even a novice like me. The only thing confusing is all firmware in my sofrtware shows as chameleon 1.0 even if I use the old one which i think is stock or the new one from here.

bogiton, 2019-02-14:
Yeah, I agree that the identical version is a bit confusing, but the reason we didn't change it was for the newer firmware to be compatible with the stock (chinese) GUI. When we drop support for that, and only support the newer GUI, then we can safely go on with some proper versioning. :)

iceman1001, 2019-02-15:
Yup,   with the introduction of multilanguage and a chinese translation the GUI is well prepared for a swap of firmware version/naming.   However we don't work active with the firmware.  I have no time for it but so much things could be done with it.