iceman1001/ChameleonMini-rebooted #72: Change IC type?

dealaddict, 2019-02-15:
I have a 1x1 clone of my tag working perfectly on Chameleon. But it's showing my IC type as Ntag 203 instead of 213. I tried to verify all the bits and bytes in my hex editor from the dump file and everything is identical...

Is this normal or there should be some section where I can modify my chameleon ntag type?

bogiton, 2019-02-16:
I'm not really sure what are the differences between 203 and 213, but I guess it all comes down to the (hard-coded) communication responses of the Chameleon, where there is no support for NTAG yet, only for ULEV1.
My guess is that probably your reader issues a GET_VERSION command and either it gets no answer or it gets back the hard-coded value of ULEV1, so it just drops to 203.
That said, it would be nice to also add support for NTAG.

iceman1001, 2019-02-16:
yeah,  you would need to modify the get_Version command response.

closed 2019-02-21 by dealaddict