iceman1001/ChameleonMini-rebooted #75: New firmware can't sniff password

libin-ka, 2019-02-25:
The latest firmware cannot sniff the password. When the firmware is sniffed, the corresponding indicator light will flash and the new firmware will be lit when the sniffing light is on.

iceman1001, 2019-02-25:
Since its not sniffing passwords, but rather sniffing Mifare authentications online and step two is a offline recovery phase of key.  What did you do to?  Screenshots from software you are using?

libin-ka, 2019-02-25:
The same read head, that is, the latest firmware can not sniff out the password, the factory-supplied firmware sniffing password can sniff the password.

iceman1001, 2019-02-25:
see previous answer

bogiton, 2019-02-25:
@libin-ka What kind of password you want to sniff? Mifare Classic? Are you using the Chameleon mini rebooted GUI?

libin-ka, 2019-02-25:
> @libin-ka What kind of password you want to sniff? Mifare Classic? Are you using the Chameleon mini rebooted GUI?
chameleon Mini GUI -v1.20.14

iceman1001, 2019-02-26:
What are you findings @bogiton  ?  You mention that MF_DETECTION  and PASSWORD sniffing didn't work??

bogiton, 2019-02-27:
I am not sure what is happening with MF_DETECTION (or my device)... As soon as I set the configuration of a slot to that, the pm3 cannot select the card... I tested it with multiple combinations of firmware/GUI. Unfortunately, only the stock firmware worked for me. Anyone who has used this functionality with our firmware?
On the other hand, PWDMY works fine in the latest firmware.

iceman1001, 2019-03-07:
did you fix this one @bogiton ?

bogiton, 2019-03-07:
Should be the same as in issue #44 
Maybe @libin-ka can confirm if it works with the new firmware?

libin-ka, 2019-03-08:
[image]   device disconnect

libin-ka, 2019-03-08:
[image]  v1.2.0.15   
Can connect, can sniff out the password

bogiton, 2019-03-08:
Yeah, the old GUI is no longer supported in the newer firmware.
So, all good with the new one. Great! :)

iceman1001, 2019-03-08:
GUI v1.2.15 works.  ok.  Time to close!

closed 2019-03-08 by iceman1001