iceman1001/ChameleonMini-rebooted #76: SAKMY set to 20 does not work

rgrunbla, 2019-02-26:
Hi,
I'm trying the new features (SAK and ATQA settinggs). ATQA is working as expected (it seems), but for the SAK, setting it to 20 stops the chameleon mini (rev e) from working. Is that expected? The SAK seems updated, but an external reader does not succeed in reading the chameleon mini, while it works with the SAK to 19 for example.
```
SAKMY=20
100:OK
SAKMY?
101:OK WITH TEXT
20
```
```
% sudo nfc-list
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
%
```

When set to 19:

```
% sudo nfc-list
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 12  34  
       UID (NFCID1): 12  34  56  78  
      SAK (SEL_RES): 19  
```

iceman1001, 2019-02-26:
Do you randomly set SAK and ask why a reader think the anticollision phase failed?

rgrunbla, 2019-02-27:
No, I don't randomly set it. I set it to what is read by a acr122u when scanning a mifare desfire card.

iceman1001, 2019-02-27:
did you know that ATQA/SAK works together?

If you want to add a check that verifies all possible valid combos of ATQA/SAK and add it into the firmware,  feel free to contribute!

rgrunbla, 2019-02-27:
That's why I tested ATQA/SAK that should works together, as listed on the http://nfc-tools.org/index.php/ISO14443A.

For example, 03 44 ATQA + 20 SAK does not work, but it should (or at least, some real world cards are using this).

iceman1001, 2019-02-27:
Your original post doesn't show any signs on which ATQA values you used, more than 12 34.
In order to figure out if this is a bug or not,  traces of communications and steps / values you have done, set.

rgrunbla, 2019-02-27:
So

1. I cloned this repo, compiled the firmware, and put it on a chameleon mini rev e.
2. I executed these commands:

```
HELPMY
101:OK WITH TEXT
VERSIONMY,CONFIGMY,UIDMY,ATQAMY,SAKMY,READONLYMY,UPLOADMY,DOWNLOADMY,RESETMY,UPGRADEMY,MEMSIZEMY,UIDSIZEMY,BUTTONMY,BUTTON_LONGMY,SETTINGMY,CLEARMY,HELPMY,RSSIMY,PWDMY,DETECTIONMY,SPI_FLASHINFOMY
VERSIONMY?
101:OK WITH TEXT
Chameleon Mini Rebooted v1.2 (Iceman)
CONFIGMY
101:OK WITH TEXT
CLOSED,MF_ULTRALIGHT,MF_ULTRALIGHT_EV1_80B,MF_ULTRALIGHT_EV1_164B,MF_CLASSIC_1K,MF_CLASSIC_1K_7B,MF_CLASSIC_4K,MF_DETECTION
CONFIGMY=MF_ULTRALIGHT
100:OK
UIDMY=11223344112233
100:OK
UIDMY?
101:OK WITH TEXT
11223344112233
ATQAMY=0344
100:OK
ATQAMY?
101:OK WITH TEXT
0344
SAKMY=20
100:OK
SAKMY?
101:OK WITH TEXT
20
SAKMY?
101:OK WITH TEXT
20

```

Using nfc-list -v, this is what I get:

```
nfc-list uses libnfc 1.7.1
NFC device: ACS / ACR122U PICC Interface opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 03  44
* UID size: double
* bit frame anticollision supported
       UID (NFCID1): 11  22  33  44  11  22  33
      SAK (SEL_RES): 00
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
Other possible matches based on ATQA & SAK values:
* Unknown card, sorry
0 Felica (212 kbps) passive target(s) found.
0 Felica (424 kbps) passive target(s) found.
0 ISO14443B passive target(s) found.
0 ISO14443B' passive target(s) found.
0 ISO14443B-2 ST SRx passive target(s) found.
0 ISO14443B-2 ASK CTx passive target(s) found.
0 ISO14443A-3 Jewel passive target(s) found.
0 ISO14443A-2 NFC Barcode passive target(s) found.

```

Tell me if you need more information.

bogiton, 2019-02-27:
I think it is working ok. When I set it to the default MF_CLASSIC_1K configuration (with SAK = 08) from the GUI, I get the following output from pm3:
```
pm3 --> hf 14a info
 UID : 11 22 33 44
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1
```
Then I issue the `SAKMY=88` command from the serial terminal and I get the following from pm3, as expected:
```
pm3 --> hf 14a info
 UID : 11 22 33 44
ATQA : 00 04
 SAK : 88 [2]
TYPE : Infineon MIFARE CLASSIC 1K
```
Please correct me if I'm wrong, but I think that the Mifare Desfire configuration you are trying (with ATQA 0344 and SAK 20) requires different handling from the firmware, which we don't support yet.

rgrunbla, 2019-02-27:
> Please correct me if I'm wrong, but I think that the Mifare Desfire configuration you are trying (with ATQA 0344 and SAK 20) requires different handling from the firmware, which we don't support yet.

Do you have an idea where to start? I believe the ISO14443ASelect should work with no regard for the SAKValue, but that might be elsewhere.

bogiton, 2019-02-27:
I guess you'll have to wait until we add support for MF_DESFIRE :)

rgrunbla, 2019-02-27:
Yeah, but I mean, cannot I get a sak equal to 20 even if i'm not using MF_DESFIRE, but only MF_ULTRALIGHT ? That seems weird to me, as 0x20 is not really bigger than 0x88 for example. Why does it fails?

iceman1001, 2019-08-14:
close because of inactivity

closed 2019-08-14 by iceman1001