iceman1001/ChameleonMini-rebooted #77: Reader attack

ictmehrdad, 2019-03-01:
Hi
I have some questions:
1- Is it possible to have a reader Attack for the other RFID card!?I mean Ultralight and DESFire and etc.
2- where Can I find this version of software that you introduced in the video!?I can’t find it! I mean the version that have a fast calculator bottom to find a key. The video that exsit in lab401 site
3- why proxmark 3 RDV4 doesn’t have a battery!? how can we use it as an stand alone!?
4- Can I have a same attack with RDV4!?(I mean reader attack)

bogiton, 2019-03-01:
Hi
1- For Ultralight there is the PWDMY command. DESFire is not supported yet.
2- Try https://github.com/iceman1001/ChameleonMini-rebootedGUI
3- This is a Chameleon-Mini RevE rebooted repo, but you can use a power-bank.
4- This is a Chameleon-Mini RevE rebooted repo, but I'm sure you can do much more with the RDV4.

ictmehrdad, 2019-03-01:
Thank you for your good answer 
About your number 4 answer:
Could you please tell some the more thing that I can do with RDV4!?
Thanks

bogiton, 2019-03-01:
Maybe I wasn't very clear. This is a Chameleon-Mini RevE rebooted repository. You should head to https://github.com/RfidResearchGroup/proxmark3 which is the official repo for RDV4. Since proxmark is not just an emulator (like Chameleon RevE rebooted), you can do more stuff.

ictmehrdad, 2019-03-01:
Thank you
I just buy Chameleon rebooted too
But I don’t know How Can exactly do with that(refer to your answer)
Could you please tell me the topics that we can able to do with Rebooted!?

closed 2019-03-01 by iceman1001